mortbopet/Ripes #3: Error ?

SenhorN, 2018-08-29:
error in:

.data
str: .string "841"
a0: .word 0 #Contador

**.text
addi t5,zero,48
addi t6,zero,57
addi t3,zero,0
la t1,str**
length:
lb t2,0(t1)
blt t2,t5,fimContagem # se t2<48,(48 representa 0, em decimal, ASCII) vá para fimContagem
blt t6,t2,fimContagem # se 57<t2 , vá para fimContagem (57 representa 9 em decimal, tabela ASCII)
addi t3,t3,1 #incrementando o contador do tamanho da palavra

addi t1,t1,1
beq x0,x0,length

fimContagem:

but, if change to this new code, it's okay:

.data
str: .string "841"
a0: .word 0 #Contador

.text
**la t1,str
addi t5,zero,48
addi t6,zero,57
addi t3,zero,0**
length:
lb t2,0(t1)
blt t2,t5,fimContagem # se t2<48,(48 representa 0, em decimal, ASCII) vá para fimContagem
blt t6,t2,fimContagem # se 57<t2 , vá para fimContagem (57 representa 9 em decimal, tabela ASCII)
addi t3,t3,1 #incrementando o contador do tamanho da palavra

addi t1,t1,1
beq x0,x0,length

fimContagem:

mortbopet, 2018-08-29:
@SenhorN When you mention that there is an error in the first code snippet, what error are you referring to?

SenhorN, 2018-08-29:
**lb t2,0(t1)** # don't work.

mortbopet, 2018-08-31:
@SenhorN fixed in 65f9cc542069d7bf77da27f89826c97ba8503cf8 
thank you very much for reporting the issue.

closed 2018-08-31 by mortbopet