mortbopet/Ripes #8: Incorrect behavior on recursive function calls

aricneto, 2018-09-15:
There's a rather critical bug in the simulator related to recursive function calls. I haven't been able to pinpoint exactly _where_ the problem lies, but I'd guess it's got something to do with instructions that manipulate the return address register (`jal`, `jalr`, ...).

The following code implements a function for recursive factorial calculation. It's almost verbatim to an example shown in the Computer Organization and Design: RISC-V Edition book:

```
main:
        addi     a0, zero, 7 # function argument (7)
        jal      ra, fact 
        jalr     x0, x1, 0

fact:
        addi     sp, sp, -16 
        sw       ra, 8(sp) 
        sw       a0, 0(sp)  
        addi     t0, a0, -1
        bge      t0, zero, nfact

        addi     a0, zero, 1
        addi     sp, sp, 16 
        jalr     x0, x1, 0
                                                     
nfact:
        addi     a0, a0, -1 
        jal      ra, fact 

        addi     t1, a0, 0 
        lw       a0, 0(sp) 
        lw       ra, 8(sp) 
        addi     sp, sp, 16 

        mul      a0, a0, t1
        jalr     x0, x1, 0 
```
The register `a0` is the function argument _and_ the return register. It starts out as 7, so at the end of the program it should read 5040, which is 7 factorial.

If you run it in the RIPES simulator, it'll end as soon as `a0` equals 0. The program will step through the final lines before the return, add 1 to `a0`, 16 to `sp`, and end. The final value of `a0` will always be 1, regardless of the input.

However, if you compile the same code using the official [RISC-V compiler toolchain](https://github.com/riscv/riscv-tools), simulate it with the [ISA simulator](https://github.com/riscv/riscv-isa-sim) and step through it with the included GDB toolkit, the program works just fine, and outputs 5040 on `a0` when it ends, as shown by the register dump below:

```
zero: 0x0000000000000000  ra  : 0x0000000000010080  sp  : 0x000000007f7e9b50  gp  : 0x0000000000000000  
tp  : 0x0000000000000000  t0  : 0xffffffffffffffff  t1  : 0x00000000000002d0  t2  : 0x0000000000000000  
s0  : 0x0000000000000000  s1  : 0x0000000000000000  a0  : 0x00000000000013b0  a1  : 0x0000000000000000  
a2  : 0x0000000000000000  a3  : 0x0000000000000000  a4  : 0x0000000000000000  a5  : 0x0000000000000000 
```

If it helps, I also used a C compiler to compile a simple recursive factorial function [into RISC-V ASM](https://gist.github.com/aricneto/a23c174d78a120332f28a5cf52d1109c), and it had the exact same issue.

mortbopet, 2018-09-15:
Thank you for finding the issue and writing a comprehensive description.

I located the issue as being related to how the pipeline should initiate its finishing sequence when it is out of instructions to execute. Seemed like i determined that the simulator could finish when the PC in the IF stage was creater than the .text size. This is ofcourse wrong, since - as in this case - if the final instruction is a control flow instruction, then this instruction must be allowed to propagate to _at least_ the ID stage.

I've modified the program slightly, to the following:
```
.data
argument: .word 7 # Number to find the factorial value of

.text
main:
        lw       a0, argument
        jal      ra, fact
        jal      zero, end            # <-- Modified

fact:
        addi     sp, sp, -16 
        sw       ra, 8(sp) 
        sw       a0, 0(sp)  
        addi     t0, a0, -1
        bge      t0, zero, nfact

        addi     a0, zero, 1
        addi     sp, sp, 16 
        jalr     x0, x1, 0
                                                     
nfact:
        addi     a0, a0, -1 
        jal      ra, fact 

        addi     t1, a0, 0 
        lw       a0, 0(sp) 
        lw       ra, 8(sp) 
        addi     sp, sp, 16 

        mul      a0, a0, t1
        jalr     x0, x1, 0 
end:                                      # <-- added
```
The simulator will go into an infinite loop running the original code fragment, since `ra` will at that point refer to the same `jalr` instruction which is meant to branch out of main. Ripes currently does not have any detection for inifnite loops (or parallel execution - the simulator runs in the GUI thread and thus stuff like this will block the gui). 

Instead, i've modified it so that the simulator jumps to the end of the program, and from there, stops.

mortbopet, 2018-09-15:
Added your code as an example that is bundled with the program. Again, thank you very much for taking your time to report this issue.

closed 2018-09-15 by mortbopet