mortbopet/Ripes #17: Add instruction formats to assembly error tooltip

mortbopet, 2018-10-08:
For instance, if a user tries to write `jalr x0 0(x1)` the tooltip should also state that `jalr` has the format
`jalr [rd] [rs] [imm]`.

jayhua267, 2021-01-11:
hello @mortbopet 
I was trying to implement Ripes through this issues.
After collecting some RISC base instruction, the tooltip will display like this picture
I am looking forward to implementing this with some pseudo instruction or maybe other tooltip
[image]

mortbopet, 2021-01-11:
Hi @jayhua267 
Very cool! It might be a good idea to take a look at the work on the [assembler branch](https://github.com/mortbopet/Ripes/tree/assembler). Soon, the assembler/disassembler (+assembler error handling) infrastructure is due for a change, based on what is in that branch.
Hopefully, you will find that the code is a lot cleaner than what you've probably encountered on the master branch, wrt. error handling and syntax highlighting.

To get you started, the most important points are:
- [edittab.cpp:215](https://github.com/mortbopet/Ripes/blob/assembler/src/edittab.cpp#L215) Main assemble routine, triggered upon changes in the code editor. The `m_sourceerrors` is shared with the `codeeditor`. As such, any errors (generated by the assembler) will be implicitly shared to the codeeditor.
- [codeeditor.cpp:125](https://github.com/mortbopet/Ripes/blob/assembler/src/editor/codeeditor.cpp#L125) sets the error generated by the assembler on the given source code line
- And there is the [assembler:245](https://github.com/mortbopet/Ripes/blob/assembler/src/assembler/assembler.h#L245) which is the main part of this. At any point during assembly, each of the passes can either run successfully, or return with some error. If an error is returned by the pass, this error is used in the return call ([see this ](https://github.com/mortbopet/Ripes/blob/assembler/src/assembler/assembler.h#L23)). Likewise, internally in a pass, we may gather errors from operations, ie. [executing tokenization of an input line](https://github.com/mortbopet/Ripes/blob/assembler/src/assembler/assembler.h#L356) using [this](https://github.com/mortbopet/Ripes/blob/assembler/src/assembler/assembler.h#L38) macro. Actual instruction errors are then generated based on their representation to the assembler. For this, you'd need to take a look at [how instructions are assembled and can throw errors](https://github.com/mortbopet/Ripes/blob/assembler/src/assembler/instruction.h#L263) as well as [how the actual assembler is defined](https://github.com/mortbopet/Ripes/blob/assembler/src/assembler/rv32i_assembler.cpp).

NCKUMaxSnake, 2021-01-12:
Hi, @mortbopet 
This is @jayhua267 classmate, The Ripes is very strong and cooool.
And I have a question want to know.
- Can I get some debug message if I add some "print" to the code?

Thank you for your reply.

mortbopet, 2021-01-12:
I am afraid i don't quite understand what you are asking about - are you referring to generating print/debug messages within ripes itself (ie. through the code editor) or when editing the source code of Ripes itself?

NCKUMaxSnake, 2021-01-12:
> I am afraid i don't quite understand what you are asking about - are you referring to generating print/debug messages within Ripes itself (ie. through the code editor) or _when editing the source code of Ripes_?

I think what I was saying is generating print/debug messages when editing the source code of Ripes.
That can help us to observe the code flow.
Thanks.

mortbopet, 2021-01-12:
In that case, I'd probably recommend you to get an IDE set up (QtCreator would be ideal given that Ripes is a Qt-based application, and is what i use myself) such that you can use the built-in debugger.
For reference, please see: https://doc.qt.io/qtcreator/creator-debugging.html

sam2468sam, 2021-01-13:
Hi, @mortbopet
I'm @jayhua267 classmate.
we have a little progress like the picture below
[image]
but we think our code is not optimized yet.(Using a lot of if function in [instruction.h](https://hackmd.io/UNwy07-oQz6JjmBFOaLppQ)
To optimized our code,  we think are there parameters  about opcode type?
Or we should add some class by ourselves.

mortbopet, 2021-01-13:
The important part about the new assembler infrastructure (instruction.h) included, is that it does _not_ contain ISA-specific code. The intention is, that everything related to the ISA is specified in a specialization of the assembler, ie. [RV32I_Assembler](https://github.com/mortbopet/Ripes/blob/assembler/src/assembler/rv32i_assembler.cpp). This is done such that we in the future may add different ISAs to Ripes, and while having to do practically no work in implementing an assembler. I would recommend you to take a look at ie. [the macro for defining R-type instructions](https://github.com/mortbopet/Ripes/blob/assembler/src/assembler/rv32i_assembler.cpp#L68) and see how that corresponds to the R-type instructions in the RISC-V isa.

As you see in [rv32i_assembler](https://github.com/mortbopet/Ripes/blob/assembler/src/assembler/rv32i_assembler.cpp), we here specify the fields of each instruction, ie. _where_ the registers are defined within the instruction word itself, as well as which index it is in an assembly source code line. Similarily, immediate values can also be described through here. All of these instructions are then used to _generate_ the assembler/disassembler. 

Now, with regards to error generation. An `Instruction` has a `std::vector<std::shared_ptr<Field>> m_fields` member containing the fields defined for the instruction. `m_fields[0]` is always an `Opcode` whereas the other fields in that vector may be a `Reg` or an `Imm`. So, you can `dynamic_cast` those fields to figure out their type. After that, ie. for immediates, you can query the immediate to check its supported width etc. The only thing you _don't_ get out of this is the register naming convention `rd/rs#`. Which is something that you could take a look at how to implement in a good way (ie. this must be information specified at the ISA-level representation of the assembler).

sam2468sam, 2021-01-18:
Hi, @mortbopet
We modify our code with your opinion, but when we pull requests, we get the error message like [this].(https://travis-ci.org/github/mortbopet/Ripes/builds/754568916)
We try to fix it, but we fail. Could you take a look? We are sorry about it.

mortbopet, 2021-01-18:
Don't worry about those errors for now - they are (as far as i remember) mostly related to outdated compiler versions that doesn't fully support std::variant.

jayhua267, 2021-01-18:
Hi, @mortbopet 
According to your comment, we've just pulled another version on pull request.
And this is our result.
[image]
However, there are some suspected place that we want make sure whether the width of the [Imm] part in U-Type is true. 
(Because the [Imm] part of the U-type only has 21 bits.)
[image]

mortbopet, 2021-04-10:
Added in 0f2393a5c8a9567278dfae5944a1c1da2742daed

closed 2021-04-10 by mortbopet