mortbopet/Ripes #25: [Feature Request]: Add Continuous Save Option

khoitd1997, 2019-02-20:
Hi, Ripes is really good and thank you so much for open sourcing it to the community. I noticed that it has a save function but every time you save, you need to pick the file again, it would be nice to be able to select only once and it will keep overwriting that file and reselect again when needed. Also, it would be nice to have a similar option for the output of the assembled code where every time a new code(in Disassembled or Binary) is generated it saves to a file. I'm asking for these features since I'm taking a class where we design our processors and I have to change the assembly and grab the binary to test the processor relatively frequently. I have worked with Qt before, not specifically with Widget but I would be down to help implement this feature.

mortbopet, 2019-02-20:
Great suggestion - i see how it can be frustrating to repeatedly being prompted to "save as..." - i'll take a look at it asap.

And great idea with the storing of the disassembled/binary view next to the source code. I will probably give the user an option whether to store text-encoded binary, raw binary (suitable for loading by other simulators or onto an FPGA), or plain disassembled. I assume you must be using a small script currently, to convert the (text encoding) binary view to an actual binary file?

khoitd1997, 2019-02-21:
Right now, I'm just relying on copy and pasting the binary text into a separate .bin file. The test setup that we have right now can read the binary text file and convert that into instructions.

mortbopet, 2019-02-23:
As mentioned in the commit text, the binary file is stored as binary encoded as opposed to text-encoded. I imagine that most people would use the binary files to load into their own simulators, and as such, this data should be inserted directly into their simulators instruction memory, rather than adding an intermediate step for converting textual 1's and 0's to binary data.

Let me know what you think about the change!

khoitd1997, 2019-02-25:
Hi, thank you for the fast response, the assembly saving works fine but the binary one couldn't be read by my Vivado test bench, the text-encoded binary seems to work fine though so I think having an option to select either version would be nice.

mortbopet, 2019-02-25:
Looks like there was an issue in writing the binary files; fixed in 4f3d1924f18e0defd100318d59f13acc02ce3e8b

khoitd1997, 2019-02-25:
I tested the new version again, and it doesn't seem to work. I think having it as text-encoded might be better or having the user choose the form it is saved.

mortbopet, 2019-02-26:
I do get your argument, but i think there is greater value in having in it as a flat binary file, given that:
- Users can easily get textual view of the file using standard tools (`xxd -b ${filename}.bin`)
- If executing on a simulator, users would eitherway have to transform text to an actual binary file, which is to be loaded into simulator memory.

What are the issues which you are experiencing? If you execute `xxd -b` on the binary file, is it the same contents as the binary view in Ripes? Are you loading the binary file with the correct endianness?

khoitd1997, 2019-02-26:
Hi, when I run the ``xxd -b -c 4`` command, it indeed gives the correct output but with reverse endianess, but I don't think that matters since according to the [Verilog Standard section 21.4](http://www.ece.uah.edu/~gaede/cpe526/2012%20System%20Verilog%20Language%20Reference%20Manual.pdf), the file to be used with readmemb will be a text file.

mortbopet, 2019-02-26:
I wasn't aware of that functionality in Verilog - Nifty! i get why you would prefer the text format - this has now been added in 74850e3183dc13307b86c70991003cf2f50ba3f5.
Regarding endianness, the binary file which is written (when writing as a flat binary file) is little endian, which is the defined endianness for RISC-V.

khoitd1997, 2019-02-26:
Thank you very much! It works really well. I will consider this issue closed. Keep up the good work!

closed 2019-02-26 by khoitd1997