mortbopet/Ripes #27: "li" not correctly expanded

adrian-azad, 2019-02-24:
Hello,
I try writing an "li" pseudo-instruction in Ripes and got an issue,  for example : 

```li t2, 0x76543210```

is expend in Ripes to  (wrong)
```
lui x7 484676 # 0x7644
addi x7 x7 528 # 0x210
```
and in gcc to (right)
```
lui t2,  484675 # 0x76543
addi t2, t2, 528 # 0x210
```

It seems coming from https://github.com/mortbopet/Ripes/blob/master/src/assembler.cpp#L436:L454 and the `+ posOffset` ... (and i don't understand why this is done...)

adrian-azad, 2019-02-24:
Ok got it, you should do `+1`sometime, (but i don't have all case in mind) for example : 
```li t2, 0x4F00```
=> 
```
lui x7 5
addi x7 x7 -256 
```

mortbopet, 2019-02-24:
The +1 offset is incorrectly calculated in Ripes currently - thanks for finding the issue!
The correct way is;
The upper 20-bit immediate should be added +1 if the lower 12 bits form a (signed) negative number. This is to ensure that when the lower 12 bits is added using `addi` to the register result of `lui`, the +1 negates the sign of the lower 12 bits.

closed 2019-02-24 by adrian-azad