mortbopet/Ripes #30: Issue if infinite loop starts accidently

anasshamoon12002, 2019-03-29:
**Infinite Loop when run in the application**

Whenever, there is an instruction that starts the loop to run infinitely many times, the program does not provide any function to stop it. 
We have to close the application using task manager.
I hope, that it gets fixed very soon.

mortbopet, 2019-03-29:
Do you have a code example?

Executing the following by pressing "run":
`
a:
j a
`
Will present a box for aborting the simulation.

anasshamoon12002, 2019-03-30:
Yup, for your code, it does show the "abort" dialogue box.

Try the following code.

a:
li a1, 10
li a0, 1
ecall
j a

anasshamoon12002, 2019-03-30:
@mortbopet 

Yup, for the above code, it does show the dialogue box.

Try the following code.

a:
li a1, 10
li a0, 1
ecall
j a

mortbopet, 2019-03-30:
Fix is available in the latest continuous release - let me know if it fixes your issies!

anasshamoon12002, 2019-03-30:
@mortbopet 

Yup, the latest release fixed the issue, and it's now working very well.

Thanks for fixing it.

anasshamoon12002, 2019-03-30:
@mortbopet 

I wanted to address another issue, that can or can not be addressed as an issue.

There is no option in 'File' to create an assembly program. I have to load the pre-existing examples to write the new code.

mortbopet, 2019-03-30:
Issues may also be used for feature requests - if you create an issue for the request, i'll add the "Enhancement" label to it.

anasshamoon12002, 2019-03-31:
@mortbopet 

Ok, I'm filing another issue on the above said missing feature.

closed 2019-03-31 by anasshamoon12002