mortbopet/Ripes #31: Variations in the instructions executed step by step or executed all once

anasshamoon12002, 2019-03-30:


anasshamoon12002, 2019-03-30:
I really appreciate your work regarding open source RISC V simulator, RIPES, but unfortunately, an issued is encountered that is to be addressed.

The step by step execution of the 'shifts' example shows variation compared to the program when run all at once.

##Shift Example when run all at once
[image]

##Shift Example with step by step execution
[image]

This issue is not only specific to this particular example, rather applies on all other examples as well.

This issue arised with the latest continuous release, on 30th Mar, 2019. I had the previous version, and that showed same number of cycle counts and instruction executed for both of the scenarios.

I hope to get that fixed.

anasshamoon12002, 2019-03-31:
@mortbopet 

Kindly, check this issue as well.

Thank you

mortbopet, 2019-03-31:
Let me know what you think! :+1:

anasshamoon12002, 2019-03-31:
@mortbopet 

Great work, bro.

I wanted to know one thing, and after this, I would close this issue.

Is Data Forwarding/bypassing technique applied already in the architecture?

mortbopet, 2019-03-31:
Yes, most of the multiplexers you see in the processor view are related to forwarding.

anasshamoon12002, 2019-03-31:
@mortbopet 

Ok, Thank you.

closed 2019-03-31 by anasshamoon12002