mortbopet/Ripes #37: ADD mode of study

Riei-Joaquim, 2019-09-03:
For use in academic teaching, it would be interesting a way where it was possible to control the conflict(hazards) resolutions active or not in the pipeline.

mortbopet, 2020-01-10:
Hello @Riei-Joaquim
I just wanted to let you know that, in the upcoming version 2.0, a major new feature is the support for multiple processor models.
I plan to release with support for three models:
- RV32IM Single cycle
- RV32IM 5 stage with **no** forwarding or hazard detection
- RV32IM 5 stage with forwarding and hazard detection (similar to the current model).

This is all in active development, and i am always looking for new collaborators (especially for the implementation of new processor models). If this sounds interesting, please refer to the [2.0 development branch](https://github.com/mortbopet/Ripes/tree/2.0-dev) or throw me a question.

Riei-Joaquim, 2020-02-15:
Hi @mortbopet 
Following the commits of the 2.0-dev branch I saw that you restructured the project around cmake leaving the standard Qt project structure aside.
I am interested in expanding the tool and adapting it to the needs of the subject. I am a monitor at my university and work with Risc-V, but with the new structure of the project some questions were not clarified and doubts arose:
- how to define a development configuration and compile exports regardless of the project to facilitate the use of the tool?

mortbopet, 2020-02-15:
If you only want to build the main application, no additional flags are required to be specified when configuring the CMake project. As such, the project can be built as any other CMake project (as explained on the front-page README).

If you by compile exports are referring to the CMake flags specific for this project, the following two exists:
- `RIPES_BUILD_TEST`: Includes the `test` subdirectory in the build
- `RIPES_ENABLE_RISCV_TESTS`: (Visible after enabling the above flag) builds RISC-V test suite. Note, that it is required to have a compatible RISC-V toolchain available if enabling this flag. A compatible toolchain can be built through the steps described [here](https://github.com/mortbopet/Ripes/wiki/(2.0)-Executing-C-programs-with-Ripes)

 In general, i can recommend working with some form of CMake GUI when exploring CMake projects and their available configurations. Personally, i do most of Ripes development using QtCreator, which has excellent CMake integration.

mortbopet, 2020-04-20:
Hello @Riei-Joaquim, I am doing some issue cleanup;
with regards to the initial issue which you raised, is this sufficiently addressed in the current status of the program?
If so, please close this issue. If not, please provide further details.

Riei-Joaquim, 2020-05-12:
Hi @mortbopet,
A new version shows a great evolution of the tool, becoming interesting for teaching and understanding the Risc-V platform, with the software architecture designed in this version, it is more accessible additions and modifications for new demands.
Good job!

closed 2020-05-12 by Riei-Joaquim