mortbopet/Ripes #39: improvements

idpan, 2019-09-08:
hi, congrats cause you developed the only visual simulation for riscv. i'm trying to create  a riscv  simulator for teaching purposes and i find this very promising. would you be interesting to communicate with me to see if there is a way to help each other improve ripes so it can be used as a tool in advanced architecture classes?

mortbopet, 2019-09-08:
Sure!

Although i haven't had time to actively develop the project in a while, [VSRTL](https://github.com/mortbopet/VSRTL) was intended to be the follow-up project to Ripes. The intention is to provide a library/framework for generalizing all of the ugly code in Ripes (the connection between the simulator and visualization, proper algorithm for circuit clocking etc.).
Having that, i intended to do things such as follows:
- Different processor implementations; ie. single cycle, 3 stage and 5 stage
- Different versions of hazard detection, forwarding, branch logic etc.

However, i do not foresee that i will have much time for actively developing the library (or major additions/modifications of Ripes) in the near future - but i would gladly be involved in discussions regarding the topic.

idpan, 2019-09-10:
It would be great to add branch prediction with the ability to choose in what stage it will happen and forwarding. Also a further add could be cache memory. I thing with these improvements it could be a great first step for a tool not only for riscv amateurs to get started but for advances architecture students too.

mortbopet, 2021-01-02:
moved to discussions page.

closed 2021-01-02 by mortbopet