mortbopet/Ripes #40: QRegularExpression: No such file or directory

Dantali0n, 2019-10-05:
When I try to compile on Ubuntu 18.04 I get the error that QRegularExpression is not found. I have both qt 5.11 and 5.13 installed.

I run qtchooser with qmake -qt to set the version of qt to one that contains QRegularExpression but still QT tries to use qt4 and thus fails to find the QRegularExpression.

mortbopet, 2019-10-05:
QRegularExpression is part of the [Qt Core classes](https://doc.qt.io/qt-5/qtcore-module.html) so if you are having include errors for QRegularExpression i would think that you have an issue with your Qt setup.

Are you able to compile any of the examples bundled by Qt? (Ie. the Qt Regular Expression example: https://doc.qt.io/qt-5/qtwidgets-tools-regularexpression-example.html )

Dantali0n, 2019-10-06:
yes the compilation of this example works as expected when I provide the -qt=5.13 parameter to qmake. I suspect your project file is somehow not correctly picking up the specified qt toolchain as I can see the gcc parameters refer to qt4.

mortbopet, 2019-10-06:
AFAIK, selecting the Qt distribution/toolchain should not be up to the Qt .pro file, but instead the QMake version which is executing the given project file.

Does the error persist if run the .pro file explicitly using `qmake` for either your 5.11 or 5.13 install and not through qtchooser?

Dantali0n, 2019-10-06:
Yes the errors persist using `qmake` both with `qmake -qt=5.11` and with `qmake -qt=5.13` while the regularexpression example works. Sorry for the misunderstanding `qtchooser` is simply to list the available qt toolchains.

mortbopet, 2019-10-06:
I am unfortunately not able to reproduce the error. The last thing that i can think about is whether your QMake is not pulling in the Qt core libraries through the dependency on the widget library.

Does applying the attached patch fix the issue?

[0001-add-core-to-.pro-config.patch.txt](https://github.com/mortbopet/Ripes/files/3694497/0001-add-core-to-.pro-config.patch.txt)

Dantali0n, 2019-10-06:
Unfortunately, no this does not fix it. It might have something to do with the multiple project files as checking out v1.0.3 resolves the incorrect decision to choose the qt4 toolchain.

With the current version I execute

```
$ qmake --version
QMake version 3.1
Using Qt version 5.13.1 in /home/XXX/Programs/qt-5.13.1/5.13.1/gcc_64/lib
```

Yet when I do make

```
$ make
cd src/ && ( test -e Makefile || /home/XXX/Programs/qt-5.13.1/5.13.1/gcc_64/bin/qmake -o Makefile /media/XXX/Data/projects/contribute/riscv/Ripes/src/src.pro ) && make -f Makefile 
make[1]: Entering directory '/media/XXX/Data/projects/contribute/riscv/Ripes/src'
Makefile:345: warning: overriding recipe for target 'moc_addressdialog.cpp'
Makefile:342: warning: ignoring old recipe for target 'moc_addressdialog.cpp'
g++ -c -m64 -pipe -O2 -fPIC -Wall -W -D_REENTRANT -DQT_NO_PRINTER -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtSvg -I/usr/include/qt4 -I. -Igraphics -I../external/fancytabbar -I. -I. -o parser.o parser.cpp
```

Clearly the inclusion of qt4 is wrong as it should use the specified version of qt5. Setting the QT_SELECT environment variable does not help either.

Dantali0n, 2019-10-06:
If you make sure the first version of QT found in the $PATH is the correct one it works. Still I do not think this should be considered a solution as the project should correctly pick up on the specified version of qt and use that.

Dantali0n, 2019-10-08:
Deleting the entire repo and downloading it again. Seems qmake caching is aggressive preserving cache even when the value for the -qt= parameter changes.

closed 2019-10-08 by Dantali0n