mortbopet/Ripes #42: Error with li pseudo instruction

jlusiardi, 2019-11-08:
Hi,

I think there is an issue with either the code that replaces the li pseudo instruction with lui and addi or one of this instructions is buggy. 

To reproduce, enter this source code:
```
li x1, 1193046
```
The assembled output is:
```
lui x1 292
addi x1 x1 1110
```
Executing this code, I don't see the expected value of 1193046 in x1 but 1197142.

But, if I execute the following code, the result in x2 is correct.
```
lui x2, 292
addi x2, x2, 1110
```

Regards
Joachim

mortbopet, 2019-11-09:
I suspect that you might be running an outdated version of Ripes. 

`li x1, 1193046` in the available continuous build assembles as
```
lui x1 291
addi x1 x1 1110
```

and executes to `x1=0x00123456`.

jlusiardi, 2019-11-10:
Fixed in continuous with hash d3ac2c2. Thanks!

closed 2019-11-10 by jlusiardi