mortbopet/Ripes #44: Feature request

schoeberl, 2019-11-27:
You can read binary files, but save sould also be nice. That could help to use Ripes as the golden model for their own ISA simulation implementation.

mortbopet, 2019-11-27:
This should already be possible. When saving in Ripes, up to three different files will be saved, as specified by the user:
- Source file
- Disassembled file
- Binary file (saved as text or as a flat binary file).
Then, once the user selects "Save As" or "Save Files As", all of the selected files will be stored, with each filename suffixed by the file type.

[image]

mortbopet, 2019-11-27:
Indeed! Apart from a couple of new features, there's been quite a few bug fixes since then.