mortbopet/Ripes #49: CI integration

ashriram, 2020-04-14:
Is there a way to run it from the command line for batch processing?
Is there a way to dump the stage view to a file?

mortbopet, 2020-04-15:
Hello @ashriram 
> Is there a way to run it from the command line for batch processing?

Currently, no. However i would expect this to be fairly straight forward to implement. This implementation would most likely follow the same steps as done in the (at this point, only) unit test file. These unit tests runs the processors without any graphical interfacing - ie. solely through the VSRTL Core layer.
https://github.com/mortbopet/Ripes/blob/master/test/tst_riscv.cpp
The general flow of the unit test setup is:
- [An assembly file is compiled and its .text segment is extracted to a separate flat binary file](https://github.com/mortbopet/Ripes/blob/master/test/tst_riscv.cpp#L178)
- [The processor is made to load this flat binary file when requesting a program reload](https://github.com/mortbopet/Ripes/blob/master/test/tst_riscv.cpp#L184)
- [The processorhandler is set to select the given processor currently being tested](https://github.com/mortbopet/Ripes/blob/master/test/tst_riscv.cpp#L188)
- [We execute the processor until some termination constraint is met](https://github.com/mortbopet/Ripes/blob/master/test/tst_riscv.cpp#L192)

Hence, a command-line wrapper could probably reuse most, if not all of the code, presented there.

> Is there a way to dump the stage view to a file?

The closest thing to this is the stage table generated during execution, which can be copied as tab-separated, and inserted in a separate file (spreadsheet or others...).

If you would like to have this information in some other format, i would suggest adding some additional code [here](https://github.com/mortbopet/Ripes/blob/master/src/stagetablemodel.cpp#L83), in the StageTableModel, where you will be able to access the stageInfo. The stageinfo is gathered for each cycle of the execution, and contains information for each stage wrt. the address currently loaded into each stage.

ashriram, 2020-04-15:
Ok. Thanks. I will submit a pull request

mortbopet, 2021-04-10:
@ashriram Is this still an issue? If not, please close the issue.

closed 2021-05-21 by mortbopet