mortbopet/Ripes #50: Ripes build succeeds, but app crashes on catalina

ashriram, 2020-04-15:
How to replicate.
On catalina
mkdir -p Qt/5.13.0
tar -xJ -C Qt/5.13.0 -f Qt-5.13.0-mac.tar.xz
export QT_ROOT/Qt/5.13.0/clang_64
export PATH=$QT_ROOT/bin:$PATH
export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig:$PKG_CONFIG_PATH
mkdir build
cd build
cmake -GNinja ../
open Ripes.app

mortbopet, 2020-04-15:
Apart from the CI macOS instance (High Sierra 10.13) I unfortunately do not have access to other macOS machines. Is it possible to a more elaborate error message?
What is the status if you build and execute it in debug mode?

mortbopet, 2021-04-10:
@ashriram Is this still an issue? If not, please close the issue.

ashriram, 2021-04-11:
Some of the font renderings are bad; but the build is ok. 
With Big Sur replacing Catalina closing this issue for now.

closed 2021-04-11 by ashriram