mortbopet/Ripes #51: Can't see full entries in Address column

jnoche, 2020-04-19:
I am currently using Ripes version v2.0.1-1-g9578aef on an x64-based PC running Microsoft Windows 10 Home Single Language version 10.0.18363 Build 18363.  I am using it to teach a Computer Architecture course and I am very grateful that you have made the software freely available.

[image]

I cannot see the full entries for the Address column in the Memory tab.  I've tried changing the display type and right-clicking and double-left-clicking on the cells.  I cannot change the width of the column using my mouse.

Is there a way to see the full entries for the Address column in the Memory tab?

Thank you very much for your time.

jnoche, 2020-04-19:
Even though I can't change the width of the Address column, I am able to change the width of the other columns.  If I hover my mouse pointer over the right side of the Word column and reduce its width, the width of the Address column increases.  It seems that the sum of the widths of the Address and Word columns are set to some constant.

[image]

jnoche, 2020-04-19:
It seems that the sum of the widths of _all_ the columns are set to some constant.  I was finally able to display the full entries for all the columns by making the widths of all the other columns smaller.

[image]

mortbopet, 2020-04-19:
Unfortunately, the default sizing of that window is a fixed size in the current 2.0 release.
This is going to be addresses in release 2.1 which should be out within the next couple of weeks.
I see that you've found somewhat of a workaround; if you find that the workaround is insufficient, i can push a hot-fix which can be available as a continuous release. Please let me know!

jnoche, 2020-04-19:
Thanks for your quick reply.  My workaround is enough for now; there is no need for a hot-fix.  Thanks again.

closed 2020-04-19 by jnoche