mortbopet/Ripes #53: bundle gcc toolchain into linux release

itviewer, 2020-04-24:
for easy of use, we can bundle the gcc toolchain into release(only for Linux?),
then add a c language code input area above assembly code, and provide some buttons like 'compie' , then automatically process the compiled code to meet the required format.

This is very convenient for learning and teaching Computer Architecture Course.

mortbopet, 2020-04-24:
Hello @itviewer 

This is something that i have been considering myself.
However, currently, i would be against bundling a toolchain along with Ripes, for the following reasons:
1. The major increase in size of the application/distributed package
2. The codebase may start to diverge and be platform-specific, given that we would probably only be able to ship compilers on some platforms.
3. In principle, I do not feel like it is the responsibility of Ripes to ship a prebuilt version of a compiler. 

However, instead of shipping a compiler along with Ripes, I'd see it more useful to be able to register a user-provided compiler with Ripes (similarly to how most IDEs work). Then, if a user has provided a path to a compatible compiler, such a C-input window may be enabled.

itviewer, 2020-04-25:
sounds good.
Is it possible to automatically build a suitable compiler(only for Linux)  through CI and then tell the user where to download the toolchain?

mortbopet, 2020-04-25:
Yep that would definitely be possible. This would simply require setting up some CI job to mirror https://github.com/riscv/riscv-gnu-toolchain and building ie. the `gcc-7-riscv64-unknown-gnu` toolchain (https://github.com/mortbopet/Ripes/wiki/Building-and-Executing-C-programs-with-Ripes).

mortbopet, 2020-04-25:
Doing a bit of digging, it seems like SiFive has already done the work for us:
https://github.com/sifive/freedom-tools/releases

Wherein we have prebuilt versions for each of the targetted platform of the required toolchain (riscv64-unknown-elf-gcc).

mortbopet, 2020-04-30:
@itviewer during the last week or so, I've been working on getting a prototype of this up an running.
It is now possible to register a compiler with ripes and select C as the input type in the editor tab. Then, everything relating to calling the compiler and loading the compiled program is handled behind-the-scenes.

You can try out this new feature in the prerelease builds i just pushed: https://github.com/mortbopet/Ripes/releases/tag/continuous-prerelease

I haven't written a guide yet to set it up, so I'll briefly explain it here:
- go to "edit->settings->editor"
- Select the path to a valid RISC-V C compiler.  You may download one at https://github.com/sifive/freedom-tools/releases . Currently, you should point to the 'riscv64-unknown-elf-gcc' within the binaries folder of the toolchain
- If everything works well the compiler path should now turn green, indicating that the compiler is valid
- Select a C example, ie. "file -> load example -> C -> ranpi.c"
- Build the example (press the build button in the toolbar or press CTRL+B)
- The program should now be visible in the disassembly tab and loaded into the simulator.

Let me know what you think!

Note: i have only tested this on my own Linux machine.

itviewer, 2020-04-30:
great!
Can it's behave more like the [Compiler Explorer](https://godbolt.org/)?
[image]
At the beginning of learning computer architecture, we introduce to students how  c language convert into assembly language.
The above picture tells us the correspondence between RISC-V instructions  and c statements very directly and concisely.

currently, our program reports a link error because there is no main function. 
If the main function is added, too much useless information appears.

mortbopet, 2020-04-30:
This is an interesting question;
While i agree that compiler explorer gives a superior view in terms of visualizing the translation between source and assembly, I think it is also important to stress that it is not a simulator.
We still need an entry point (Which is the part that compiler explorer leaves out, for good reasons), which is provided by the default crt0 which in turn is provided by the toolchain.

Now, we could write our own crt0 and link to this during compilation - and as such avoid all of the "stuff" which get's dragged in when linking using the default crt0. Essentially turning the emitted program into something akin to what we see through compiler explorer.

However, doing so would severely limit us in terms of actually being able to leverage the compiler. Say that we want to call standard library functions - some of which relies on us linking to a more elaborate crt0 as well as the standard library functions themselves. These will be fairly large functions which all may "clutter" the binary even more. But all essential for proper execution of the program.

Now, note that i _do_ agree with you, that it should be easier to navigate the program viewer in terms of analyzing the things we care about - ie. the code which we wrote ourselves.
I plan on adding symbol navigation between the source and program view. As such, you shall be able to click on a function definition in the C source, and that will move the programviewer to the given source.
Furthermore, we may highlight user-written functions in the program viewer a bit like what is done in compiler explorer, to be able to clearly discern between user-provided and compiler-provided code.

Finally, I would also say that there is value in learning what all the _other_ stuff which gets included into an executable is; and why all of this stuff is needed when we want to run a C program, as compared to running a simple assembly program.

Let me know what you think.

itviewer, 2020-05-05:
Can we make a setting option that whether it links to default crt0 or user defined one? 
and also allow user to specify other gcc parameters.

mortbopet, 2020-05-09:
@itviewer 

Seems like this can be solved simply by stating `-nostdlib` as a compiler argument.
Additional compiler arguments may be set within the settings. Then, the entry point will be the first defined function within the translation unit.
Then, from this point, it would be up to the programmer to write whatever setup and teardown code such that it makes sense to directly execute the compiled code.

[image]
[image]

itviewer, 2020-05-10:
awesome!!

closed 2020-05-10 by itviewer