mortbopet/Ripes #58: make layout suitable for laptop screen

itviewer, 2020-06-29:
I don’t know which widget restricts the minimum size, the width of the entire window exceeds the screen width and cannot be reduced. 
The laptop I used is a bit old, the resolution is 1366*768, which was popular in a few years ago.

mortbopet, 2020-06-29:
is this version 2.0.1 or on master/the current continuous release (which has the cache simulator in the memory view)?

itviewer, 2020-06-30:
I use the [Continuous build (prerelease)](https://github.com/mortbopet/Ripes/releases/tag/continuous-prerelease)

mortbopet, 2020-06-30:
So, the culprit was the memory tab wherein both the cache simulator and memory view had a rather large minimum width. fd7c6cab309c713a13680208bd7842a9a7a39f23 adjusts the MemoryViewerWidget to be able to be reduced in width.
Tested on 1280x720, Ripes is maximizable.

Please let me know if the fix is sufficient - a new prerelease has been uploaded with the changes applied.

itviewer, 2020-07-01:
the layout is good.
there seems to be a new problem for memory view, Its vertical scroll bar length does not match it's content.
[image]

Another question, the memory table cannot display all columns because the Cache view has a minimum size. should we put the memory view into the tab?

mortbopet, 2020-07-01:
> there seems to be a new problem for memory view, Its vertical scroll bar length does not match it's content.

I am a bit unsure what you mean; in "matching its content" would you here expect that there is no scrollbar? (in other words, that all memory addresses in the memory view are currently visible. In the case of your screenshot, there should be 1 additional address visible when we scroll down). If so, this behavior could be addressed to ensure that there are never so many memory addresses in the view such that a scroll bar appears.

> Another question, the memory table cannot display all columns because the Cache view has a minimum size. should we put the memory view into the tab?

I have two concerns with this:
1. It might be rather simple, but i am not a big fan of having separate tabs, wherein each tab would be rather sparsely populated (as seen in the current release version, see image below). If the majority of users were using low-resolution screens, I agree that these should be split into separate tabs. However, I would assume that the majority of users are using 1080p screens or above, and as such we should cater the design to such resolutions. And then, for users on low-resoltuion screens to avoid the situation as shown above wherein not all of the columns are visible, a user is able to collapse the cache view using the splitter between the two views, to dedicate all of the screen real-estate to the memory view (do you think the splitter should be made more visible?).

2. The cache view and memory view are rather coupled, given that when one clicks an item in the cache, the memory view automatically gets scrolled to the position of said item. I do not think this would be a welcome feature if the cache view and memory view were in two different tabs (such that when clicking an item, the tab is automatically changed).

---
[image]

itviewer, 2020-07-01:
> I am a bit unsure what you mean; in "matching its content" would you here expect that there

Maybe I didn’t understand the logic of the scrolling bar here. Let's take an example of a chrome browser window:
[image]
When I scroll through the mouse wheel, the position of the scroll bar on the right corresponds to the position of the page content. Of course, the longer the page, the shorter the size of the scroll bar. In addition, I can drag the scroll bar to scroll the page content.

> I have two concerns with this:

Before finding a more suitable layout, let's ignore the low-resoltuion screens first.

closed 2020-07-01 by itviewer