mortbopet/Ripes #61: Suggestion: Improve dark mode support

surprized, 2020-12-13:
One of the things that make Qt stand out against the alternatives is its nice support for themes. Currently, for Breeze-Dark, the UI looks like this:
[image]
[image]

The labels for each component and the data cache are kind of hard to read, but this is just a minor issue anyways.

mortbopet, 2020-12-13:
I definitely agree that this is a relevant issue to raise; currently, drawing (mainly VSRTL related) does not react to the palette which Qt announces (which afaik should contain whatever theme is set in the underlying OS).
Some work has been done in VSRTL to provide a darkmode-ish theme (although, again, this should be reworked to react to the palette). You can view the "progress" by, in the processor view, right click outside of the processor view, select `unlock`, repeat and select `drawing->darkmode`.

mortbopet, 2021-04-06:
b6bb95e7df41999c1924f0893c214a210aacea03 added improved darkmode for the processor view + added a button for toggling it.
The cache visualization is up next.

surprized, 2021-04-07:
Will update, thanks!

mortbopet, 2021-09-27:
Closing due to inactivity.

srynsh, 2022-09-29:
In Mac how can I switch from dark mode to light mode in v2.2.4?

khkred, 2022-10-07:
Hey I have the same question. How do I switch to light mode in Mac.

srynsh, 2022-10-07:
I found a temporary solution, the theme is same as the system theme to change the theme I changed the system theme.

mortbopet, 2022-10-07:
> I found a temporary solution, the theme is same as the system theme. So to change the theme I changed the system theme.

This would, unfortunately, be the only way to enforce this. Ripes currently has a fairly inconsistent use of Qt's [palette](https://doc.qt.io/qt-6/qpalette.html) system, where parts of the colors used in the program are manually specified, and other parts use the system-defaults (which leads to some issues when it comes to supporting both light and dark mode!). Contributions in this space, helping to get a more consistent look and feel of Ripes across systems and system themes, would be welcomed with open arms!

khkred, 2022-10-07:
> This would, unfortunately, be the only way to enforce this. Ripes currently has a fairly inconsistent use of Qt's [palette](https://doc.qt.io/qt-6/qpalette.html) system, where parts of the colors [...]

Sure I could be into it. I'll try creating a pull request in couple of days.

Monolitho, 2025-05-15:
Anyone here? The keyword highlight turned to underlight in dark model...

[image]

Monolitho, 2025-05-15:
Would you please add a function or option to switch the font color?