mortbopet/Ripes #62: Source tarball in releases does not contain external repositories

surprized, 2020-12-14:
I'm not sure whether this is a GitHub limitation or if they can be added later.

I am trying to learn how to use Open Build Service by packaging Ripes, however, the tarball from GitHub releases (https://github.com/mortbopet/Ripes/archive/v%{version}.tar.gz) does not contain VSRTL or ELFIO.
I should manually add the other files to solve this problem, but I wanted to know if something could be done upstream before taking further steps on my end.

Happy holidays.

mortbopet, 2020-12-14:
Hi @surprized; first of all, thank you very much for putting in the work to get Ripes working with Open Build Service!
Git submodules (which is the case for VSRTL and ELFIO) are **not** included in the tarbell that is created automatically, whenever a new release is submitted. Hence, one has to run `git submodule update --init` within the tarbell directory (or, directly when cloning the repo `git clone --recursive https://github.com/mortbopet/Ripes.git`) for the submodules to download.

I am unfamiliar with the specifics of Open Build Service, so this might be a dumb question, but does it specifically require tarbells? As in, is there an automated way for it to package up a repository straight from the git repository whenever a new release is submitted?

surprized, 2020-12-14:
For the openSUSE Build Service instance there are some rules that I am still understanding, but I'll see if I can set it up to get files from git.

From what I can tell this far, once everything is set up, the only thing that is left after every release is update the tarball and the version numbers in the spec file (or whatever is relevant to the target format), then trigger the build and it should be good to go. As a big plus, OBS has the ability to build Appimages, although as said before I am just getting started.

Here are the resources I have been sticking to so far:
https://openbuildservice.org/help/manuals/obs-user-guide/index.html
https://kamarada.github.io/en/2019/03/19/integrating-the-open-build-service-with-github/
https://docs.appimage.org/packaging-guide/hosted-services/opensuse-build-service.html

surprized, 2020-12-15:
Seems like I got rpm files up and running :)  ---- pending: testing -----
I might take a shot at appimage files, maybe
I can confirm OBS supports fetching from git, see the [_service example file](https://build.opensuse.org/package/view_file/home:obsgeeko/my-first-obs-package/_service?expand=1). Anyways, I made a script to automate the whole update process, so as soon as I get a GItHub notification I run the script and the whole thing updates itself. (and I stil have to test the script... hehe)

The files are available at: https://build.opensuse.org/package/show/home:andresbs/ripes

mortbopet, 2020-12-15:
Great progress!

If it is of relevance, i rely pretty much solely on [linuxdeployqt](https://github.com/probonopd/linuxdeployqt) for packaging the binary (and its dependencies) into an AppImage file (see https://github.com/mortbopet/Ripes/blob/master/scripts/linux/travisbuild.sh).

surprized, 2020-12-15:
That seems to make your life a lot easier as well.
In a couple weeks I might take a look at this to see what should be packaged in the appimage.
The file I have to make should look [something like this](https://build.opensuse.org/package/view_file/home:probono/QtQuickApp/appimage.yml?expand=1), and it certainly is _very familiar_ to your build setup.

For now, the targets I have working properly are:
Fedora 33 (x86_64)
openSUSE Leap (x86_64)
openSUSE Tumbleweed (x86_64,aarch64,i586)

I have only tested the Leap RPM, but I expect the others to work just fine except for i586 (I'd have to undust my i386 Pentium 4, which is the closest thing I have). Right now I am sorting some issues with the Fedora build, but it should be ready in a couple hours.
If you want maintainer access to the repo let me know!

The builds are published at https://download.opensuse.org/repositories/home:/andresbs/

mortbopet, 2020-12-15:
Awesome - it's a good step forwards to now have Ripes available through some package managers. Please do keep me updated on the appimage progress.

surprized, 2020-12-15:
Alright!

closed 2020-12-15 by surprized