mortbopet/Ripes #63: Building and packaging Ripes using Open Build Service

surprized, 2020-12-15:
I have been learning how to use Open Build Service (OBS) and decided to use Ripes as my first attempt. I managed make builds for Tumbleweed, Leap and Fedora 33. They are published at [home:andresbs](https://download.opensuse.org/repositories/home:/andresbs/).

OBS can fetch files from git repos and build automatically from there, but I decided to instead make a script that clones the latest release, prepares the complete tarball for the build part and modifies the spec file accordingly.

It can also [make Appimages](https://build.opensuse.org/package/view_file/home:probono/QtQuickApp/appimage.yml?expand=1) in a process similar to the one currently used for Ripes. I will attempt this once I get RPMs working fine.

I am still sorting out issues with the .desktop file for Leap (certain openSUSE conventions), but Tumbleweed and Fedora RPMs seem to build without problems. My Fedora 33 install needs to be updated and I am currently with limited bandwith resources, I'd be happy to know if someone can test that RPM.

Either way, it should be possible to open Ripes from the terminal directly by just running "Ripes".

(see #62)

surprized, 2020-12-15:
RPMs are fully working now. Shortcut included.
Now I am trying to figure out how to build the Appimage.

surprized, 2020-12-16:
I'll leave this here in case Google, DDG, or whatever happens to find this information:
Adding the Appimage target by going to OBS (I'm on the beta) -> Repositoires ->Add from a Distribution uses a repo targeting Leap 42.3, which reached EOL, idk, long ago?
The solution was to add these repos manually: to do this click "Add from a Project" and in the new one add the other repository path (use the + button that appears in the new repo):  
* OBS:AppImage:Templates:Leap:15.2/AppImage
* OBS:AppImage/toolchain.leap_15.2
Keep them in the same order as above to avoid breaking the priority.
I think I am reaaaaally close. I basically used the same travis script that is currently used.

surprized, 2020-12-17:
I got the AppImage build to work and works excellent in openSUSE Leap (KDE)
BUT, on Fedora 33 and Ubuntu (live,20.04) I get the like 15 warnings with the same message:
`qt.svg: Cannot read file ':/icons/info.svg', because: Encoding iso-8859-1 is unsupported (line 1)`
And so, the program looks like this when running:

[image]
Maybe this could be fixed by reencoding the svg files as UTF-8?
Interesetingly, the OBS AppImage build trims like 10 MB from the Realeases file. I have no idea where those savings come from.

Almost forgot to mention, I had to change to linuxdeploy with the qt plugin because linuxdeployqt wouldn't stop segfaulting...
Fedora 33 RPM build works perfectly over here.

surprized, 2020-12-17:
Tested the resulting AppImage on Fedora 33. GUI icons working fine now! If anyone can test them please let me know the outcome! [Ripes-latest-x86_64.AppImage](https://download.opensuse.org/repositories/home:/andresbs/AppImage/Ripes-latest-x86_64.AppImage)

mortbopet, 2020-12-17:
ac5e783 should fix the icon issue, so that can be removed from the `update-ripes.sh`.

> the OBS AppImage build trims like 10 MB from the Realeases file.
Is it built in Release or RelWithDebInfo mode?

As a sidenote, a large part of the file size may be attributed to the processor layout files (see https://github.com/mortbopet/Ripes/blob/master/src/processors/layouts.qrc) which are incidentally also highly compressible files. However, these are bundled with the app using the Qt resource system which itself also compress files (see https://doc.qt.io/qt-5/resources.html#compression), so i would be surprised if some 2nd level of compression managed to provide a significant size reduction after the fact.

surprized, 2020-12-17:
> ac5e783 should fix the icon issue (however i assume the script will have to stay as it is until the next release).

That's right, but thanks for letting me know! For the next release I'll remove it.

> Is it built in Release or RelWithDebInfo mode?

Release (`cmake -DCMAKE_BUILD_TYPE=Release .`)

The build script can be found at [appimage.yml](https://build.opensuse.org/package/view_file/home:andresbs/ripes/appimage.yml?expand=1). All the AppDir files must be located at `$BUILD_APPDIR`. The build log can be seen at [_log](https://build.opensuse.org/build/home:andresbs/AppImage/x86_64/ripes/_log), however, only registered users can open it, I think. [Instead, you can find it here](https://github.com/mortbopet/Ripes/files/5710222/_log.txt), if you want to check it; the last command ran at the _[707s]_ mark.

I honestly have absolutely zero idea where those savings come from, but in case you want to compare them. The builds can be compared by mounting them: `./Ripes-0-Build105.1.glibc2.14-x86_64.AppImage --appimage-mount`

surprized, 2021-02-11:
Hi, it's me again ;-;
I am trying to build from master and tweaked some things from the RPM spec file, now it builds with the following flags for Fedora and openSUSE:
```
        -DCMAKE_INSTALL_PREFIX:PATH=/usr \
        -DINCLUDE_INSTALL_DIR:PATH=/usr/include \
        -DLIB_INSTALL_DIR:PATH=/usr/lib64 \
        -DSYSCONF_INSTALL_DIR:PATH=/etc \
        -DSHARE_INSTALL_PREFIX:PATH=/usr/share \
        -DCMAKE_INSTALL_LIBDIR:PATH=/usr/lib64 \
        -DCMAKE_BUILD_TYPE=RelWithDebInfo \
        -DCMAKE_C_FLAGS="${CFLAGS:--O2 -g -m64 -fmessage-length=0 -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables} -DNDEBUG" \
        -DCMAKE_CXX_FLAGS="${CXXFLAGS:--O2 -g -m64 -fmessage-length=0 -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables} -DNDEBUG" \
        -DCMAKE_Fortran_FLAGS="${FFLAGS:--O2 -g -m64 -fmessage-length=0 -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables} -DNDEBUG" \
        -DCMAKE_EXE_LINKER_FLAGS=" -Wl,--as-needed -Wl,--no-undefined -Wl,-z,now" \
        -DCMAKE_MODULE_LINKER_FLAGS=" -Wl,--as-needed" \
        -DCMAKE_SHARED_LINKER_FLAGS=" -Wl,--as-needed -Wl,--no-undefined -Wl,-z,now" \
        -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON \
        -DBUILD_SHARED_LIBS:BOOL=ON \
        -DBUILD_STATIC_LIBS:BOOL=OFF \
        -DCMAKE_COLOR_MAKEFILE:BOOL=OFF \
        -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF \
        -DCMAKE_MODULES_INSTALL_DIR=/usr/share/cmake/Modules
```
CMake somehow links to a non existant shared object: 
```Ripes: error while loading shared libraries: libripes_lib.so: cannot open shared object file: No such file or directory```
Do you know where could the problem be?