mortbopet/Ripes #74: Assembly comments with colons cannot be correctly recognized by the editor

leepoly, 2021-03-18:
To reproduce:
- Add a comment started with # after one assembly
- Insert a colon in the comment

for example, "    la a0, str # Load the: address of the string, placed in the static data segment"

The editor pane on the left comments the entire line, including the actual "la" instruction. It seems the "the:" is recognized as a label. The assembled pane looks fine though.

mortbopet, 2021-03-18:
Thank you for reporting the issue;
There's currently some CI issues going on, so the fix won't be available in the continuous release, but if you have the project building locally, please do confirm that the issue has been fixed.

leepoly, 2021-03-21:
Thanks! Now it now works well on the latest commit.

closed 2021-03-21 by leepoly