mortbopet/Ripes #75: la and pseudo load/store cannot support very far negative offset

leepoly, 2021-03-18:
To reproduce:

- Write a label and a "la" instruction to load that label.
- Make sure that the offset of the address of the "la" instruction and the label is -2048.
- Now the assembly pane on the right can correctly assemble the pseudo "la" instruction. It shows:
```
auipc x8, 0x10000
addi x8 x8 -2048
```
- Insert one "nop" on the middle, to make sure the offset is smaller than -2048.
- Now the assembly pane on the right **incorrectly assembles** the pseudo "la" instruction. It shows:
```
auipc x8, 0x10000
addi x8 x8 2044
```
It calculates a positive offset, which is wrong.

A temporary solution

To generate correct assembles, in `src/assembler/rv32i_assembler.cpp:148`, change the "la" generation to the following
```
        "la", {RegTok, ImmTok}, PseudoExpandFunc(line) {
            return LineTokensVec{QStringList() << "auipc" << line.tokens.at(1) << QString("((%1-__address__)&0xfffff000 + (%1-__address__)&0x1000)").arg(line.tokens.at(2)),
                                 QStringList() << "addi" << line.tokens.at(1) << line.tokens.at(1) << QString(("((%1-(__address__-4))&0xfff)").arg(line.tokens.at(2))};
```

Thanks LazyJazz for pointing it out.

mortbopet, 2021-03-20:
Thank you for reporting it. This turns out to be a bit less trivial to fix than i expected.
Basically, to handle the memory-accessing pseudoinstructions correctly, it's hard to get around the fact that the assembler does not have proper support for symbol relocation, as described in https://github.com/riscv/riscv-elf-psabi-doc/blob/master/riscv-elf.md#pc-relative-symbol-addresses

So, i'll be working a bit on adding this to get us to a point where we can simply write:
```
return LineTokensVec{LineTokens() << Token("auipc") << line.tokens.at(1) << Token(line.tokens.at(2), "%pcrel_hi"),
                     LineTokens() << Token("addi") << line.tokens.at(1) << line.tokens.at(1) << Token(QString("(%1 + 4)").arg(line.tokens.at(2)), "%pcrel_lo")};

```

leepoly, 2021-03-21:
Commit 2724adf seems does not work on my laptop: I pulled this commit and try cmake and make. It says many errors like
```
         Ripes/src/assembler/rv32i_assembler.cpp:70:111: error: no matching function for call to ‘Ripes::Assembler::Opcode::Opcode(const char [5], <brace-enclosed initializer list>)’
         new Instruction(Opcode(name, {OpPart(0b0110011, 0, 6), OpPart(funct3, 12, 14), OpPart(funct7, 25, 31)})
```
My gcc is 7.5 and I did not change any cmake configs.

mortbopet, 2021-03-21:
> Commit [2724adf](https://github.com/mortbopet/Ripes/commit/2724adf5415017383648ab9f0863412b7899e159) seems does not work on my laptop: I pulled this commit and try cmake and make. It says many [...]

It's just some intermediate work, apologies, should have pointed that out. There's some commits pending for [rv32i_assembler.cpp](https://github.com/mortbopet/Ripes/blob/master/src/assembler/rv32i_assembler.cpp) that should be available soon.

mortbopet, 2021-03-21:
@leepoly 866d4b1 should hopefully have fixed it. Most of the commit is just wrapping bare string literals inside a `Token`. The relevant points are [here](https://github.com/mortbopet/Ripes/commit/866d4b1a46be5110224718af5a35dd8202f8d6bb#diff-ad49c124a82588e5a0b51668ae69d83e2d4c7ec43d05054c6c4649936d4ec3a9R111), [here](https://github.com/mortbopet/Ripes/commit/866d4b1a46be5110224718af5a35dd8202f8d6bb#diff-ad49c124a82588e5a0b51668ae69d83e2d4c7ec43d05054c6c4649936d4ec3a9L121) and [here](https://github.com/mortbopet/Ripes/commit/866d4b1a46be5110224718af5a35dd8202f8d6bb#diff-ad49c124a82588e5a0b51668ae69d83e2d4c7ec43d05054c6c4649936d4ec3a9L149).

mortbopet, 2021-04-10:
Hello @leepoly 
Are you still experiencing the error? If not, please close the issue.

closed 2021-04-10 by leepoly