mortbopet/Ripes #80: build errors on gcc 10.2.0 since -Werror commit

AppMathIsRealMath, 2021-04-06:
Since commit bdbdd0c8bf6ead2951c7f82b145e8385d7fac4b3 I've been getting build errors like this:

Scanning dependencies of target RISC-V_lib
[  9%] Linking CXX static library libvsrtl_int_lib.a
[  9%] Building CXX object src/processors/CMakeFiles/RISC-V_lib.dir/RISC-V_lib_autogen/mocs_compilation.cpp.o
cc1plus: error: ‘-Werror=Wno-bad-function-cast’: no option ‘-WWno-bad-function-cast’; did you mean ‘-Wno-bad-function-cast’?
cc1plus: error: ‘-Werror=no-unused-function’: no option ‘-Wno-unused-function’
cc1plus: error: ‘-Werror=no-unused-local-typedefs’: no option ‘-Wno-unused-local-typedefs’
make[2]: *** [src/processors/CMakeFiles/RISC-V_lib.dir/build.make:82: src/processors/CMakeFiles/RISC-V_lib.dir/RISC-V_lib_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1572: src/processors/CMakeFiles/RISC-V_lib.dir/all] Error 2

I'm using Solus Linux and current gcc version is 10.2.0

mortbopet, 2021-04-06:
Apologies for this; this should ideally have been caught by CI, but it is unfortunately currently not running, due to some issues with Travis CI.

mortbopet, 2021-04-10:
Hello @AppMathIsRealMath 
Are you still experiencing the error? If not, please close the issue.

AppMathIsRealMath, 2021-04-10:
Your commit fixed it, thanks! I'll close the issue.

closed 2021-04-10 by AppMathIsRealMath