mortbopet/Ripes #83: Windows 10 app doesn't run

trlef19, 2021-04-11:
Hello, I have an issue with the latest ripes version (actually with all the 2.x.x versions) whenever I'm trying to run it, it just loads but it never opens. I tried running in it from cmd and powershell with no luck and c++ thing is installed. Any idea? Running v2004 x64 i7 6th gen. Thanks!

mortbopet, 2021-04-11:
Hi @trlef19 
The following may give us a bit of insight into what is going wrong:
1. download WinDbg Preview from the microsoft store (https://www.microsoft.com/en-us/p/windbg-preview/9pgjgd53tn86?activetab=pivot:overviewtab)
2. start Ripes through WinDbg Preview
3. press the green "play" button once it appears
4. paste the contents of the log here

trlef19, 2021-04-11:
> Hi @trlef19
> The following may give us a bit of insight into what is going wrong:
> 1. download WinDbg Preview from the microsoft store [...]

Hey, sorry for asking, but how do I run the app through that?

mortbopet, 2021-04-11:
No worries;
`File -> Launch executable`, navigate to where Ripes is, and select `Ripes.exe` .

trlef19, 2021-04-11:
Thanks. You want that lines in "thread" category?

mortbopet, 2021-04-11:
Initially, just the things that appear in the "Command" window.

trlef19, 2021-04-11:
Hello, I replaced my andministrator name with "***" I hope that's ok. I suppose this is the output you want:
 ************* Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       srv*
Symbol search path is: srv*
Executable search path is: 
ModLoad: 00007ff7`90d30000 00007ff7`911c9000   image00007ff7`90d30000
ModLoad: 00007ffc`321d0000 00007ffc`323c5000   ntdll.dll
ModLoad: 00007ffc`30dd0000 00007ffc`30e8d000   C:\WINDOWS\System32\KERNEL32.DLL
ModLoad: 00007ffc`2faa0000 00007ffc`2fd68000   C:\WINDOWS\System32\KERNELBASE.dll
ModLoad: 00007ffc`2f910000 00007ffc`2fa10000   C:\WINDOWS\System32\ucrtbase.dll
ModLoad: 00007ffc`31a10000 00007ffc`32152000   C:\WINDOWS\System32\SHELL32.dll
ModLoad: 00007ffb`fc260000 00007ffb`fc3c4000   C:\Users\***\Downloads\Ripes-v2.1.0-win-x86_64\Qt5Charts.dll
ModLoad: 00007ffb`e23d0000 00007ffb`e2a06000   C:\Users\***\Downloads\Ripes-v2.1.0-win-x86_64\Qt5Gui.dll
ModLoad: 00007ffc`30140000 00007ffc`301dd000   C:\WINDOWS\System32\msvcp_win.dll
ModLoad: 00007ffb`e0d80000 00007ffb`e12d7000   C:\Users\***\Downloads\Ripes-v2.1.0-win-x86_64\Qt5Widgets.dll
ModLoad: 00007ffc`30230000 00007ffc`303d0000   C:\WINDOWS\System32\USER32.dll
ModLoad: 00007ffc`2fa10000 00007ffc`2fa32000   C:\WINDOWS\System32\win32u.dll
ModLoad: 00007ffc`30910000 00007ffc`3093a000   C:\WINDOWS\System32\GDI32.dll
ModLoad: 00007ffc`2fea0000 00007ffc`2ffab000   C:\WINDOWS\System32\gdi32full.dll
ModLoad: 00007ffb`d71a0000 00007ffb`d7767000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Ripes-v2.1.0-win-x86_64\Qt5Core.dll
ModLoad: 00007ffc`15f30000 00007ffc`15fc1000   C:\WINDOWS\SYSTEM32\MSVCP140.dll
ModLoad: 00007ffc`18450000 00007ffc`18469000   C:\WINDOWS\SYSTEM32\VCRUNTIME140.dll
ModLoad: 00007ffc`31200000 00007ffc`3126b000   C:\WINDOWS\System32\WS2_32.dll
ModLoad: 00007ffc`31470000 00007ffc`3159b000   C:\WINDOWS\System32\RPCRT4.dll
ModLoad: 00007ffc`310d0000 00007ffc`311fa000   C:\WINDOWS\System32\ole32.dll
ModLoad: 00007ffc`2c980000 00007ffc`2ca1e000   C:\WINDOWS\SYSTEM32\UxTheme.dll
ModLoad: 00007ffc`2d290000 00007ffc`2d2bf000   C:\WINDOWS\SYSTEM32\dwmapi.dll
ModLoad: 00007ffc`31290000 00007ffc`3132e000   C:\WINDOWS\System32\msvcrt.dll
ModLoad: 00007ffc`30a70000 00007ffc`30dc5000   C:\WINDOWS\System32\combase.dll
ModLoad: 00000203`422d0000 00000203`4236e000   C:\WINDOWS\System32\msvcrt.dll
ModLoad: 00007ffc`303e0000 00007ffc`3048c000   C:\WINDOWS\System32\ADVAPI32.dll
ModLoad: 00007ffc`30500000 00007ffc`3059b000   C:\WINDOWS\System32\sechost.dll
ModLoad: 00007ffc`19680000 00007ffc`1969d000   C:\WINDOWS\SYSTEM32\MPR.dll
ModLoad: 00007ffc`1baf0000 00007ffc`1bb08000   C:\WINDOWS\SYSTEM32\NETAPI32.dll
ModLoad: 00007ffc`2f810000 00007ffc`2f83e000   C:\WINDOWS\SYSTEM32\USERENV.dll
ModLoad: 00007ffc`281c0000 00007ffc`281ca000   C:\WINDOWS\SYSTEM32\VERSION.dll
ModLoad: 00007ffc`15f20000 00007ffc`15f2c000   C:\WINDOWS\SYSTEM32\VCRUNTIME140_1.dll
ModLoad: 00007ffc`1c050000 00007ffc`1c077000   C:\WINDOWS\SYSTEM32\WINMM.dll
ModLoad: 00007ffc`2edd0000 00007ffc`2eddc000   C:\WINDOWS\SYSTEM32\NETUTILS.DLL
ModLoad: 00007ffc`1a180000 00007ffc`1a1a8000   C:\WINDOWS\SYSTEM32\SRVCLI.DLL
ModLoad: 00007ffc`2f2c0000 00007ffc`2f2cc000   C:\WINDOWS\SYSTEM32\CRYPTBASE.DLL
(225c.bb0): Break instruction exception - code 80000003 (first chance)
ntdll!LdrpDoDebuggerBreak+0x30:
00007ffc`322a0570 cc              int     3

mortbopet, 2021-04-11:
At that point, is the "Go" button in the "Home" pane available, and have you clicked on it?

trlef19, 2021-04-11:
No, I didn't press "play". Sorry. Here is the rest:
ModLoad: 00007ffc`32160000 00007ffc`32190000   C:\WINDOWS\System32\IMM32.DLL
ModLoad: 00007ffc`2fd70000 00007ffc`2fdf0000   C:\WINDOWS\System32\bcryptPrimitives.dll
ModLoad: 00007ffc`31330000 00007ffc`313de000   C:\WINDOWS\System32\shcore.dll
ModLoad: 00007ffb`e2260000 00007ffb`e23ce000   C:\Users\***\Downloads\Ripes-v2.1.0-win-x86_64\platforms\qwindows.dll
ModLoad: 00007ffc`30940000 00007ffc`30a0d000   C:\WINDOWS\System32\OLEAUT32.dll
ModLoad: 00007ffc`2d7b0000 00007ffc`2d7c4000   C:\WINDOWS\SYSTEM32\WTSAPI32.dll
ModLoad: 00007ffc`2da30000 00007ffc`2da42000   C:\WINDOWS\SYSTEM32\kernel.appcore.dll
ModLoad: 00007ffc`2da50000 00007ffc`2e1e1000   C:\WINDOWS\SYSTEM32\windows.storage.dll
ModLoad: 00007ffc`2f380000 00007ffc`2f3ac000   C:\WINDOWS\SYSTEM32\Wldp.dll
ModLoad: 00007ffc`305a0000 00007ffc`305f5000   C:\WINDOWS\System32\shlwapi.dll
ModLoad: 00007ffc`2f850000 00007ffc`2f86f000   C:\WINDOWS\SYSTEM32\profapi.dll
ModLoad: 00007ffb`fbf80000 00007ffb`fbfa6000   C:\Users\***\Downloads\Ripes-v2.1.0-win-x86_64\styles\qwindowsvistastyle.dll
ModLoad: 00007ffc`1a790000 00007ffc`1a79d000   C:\Users\***\Downloads\Ripes-v2.1.0-win-x86_64\imageformats\qgif.dll
ModLoad: 00007ffc`18cc0000 00007ffc`18cce000   C:\Users\***\Downloads\Ripes-v2.1.0-win-x86_64\imageformats\qico.dll
ModLoad: 00007ffb`e7260000 00007ffb`e72ca000   C:\Users\***\Downloads\Ripes-v2.1.0-win-x86_64\imageformats\qjpeg.dll
ModLoad: 00007ffc`18c60000 00007ffc`18c6c000   C:\Users\***\Downloads\Ripes-v2.1.0-win-x86_64\imageformats\qsvg.dll

mortbopet, 2021-04-11:
Alright, and i presume that at that point, the application just hangs?
In the lower part of the command window, does it still say `*BUSY* Debugee is running...` ?
[image]

trlef19, 2021-04-11:
No, sorry. I ran it again. That's the output: 
ModLoad: 00007ffc`32160000 00007ffc`32190000   C:\WINDOWS\System32\IMM32.DLL
ModLoad: 00007ffc`2fd70000 00007ffc`2fdf0000   C:\WINDOWS\System32\bcryptPrimitives.dll
ModLoad: 00007ffc`31330000 00007ffc`313de000   C:\WINDOWS\System32\shcore.dll
ModLoad: 00007ffb`e6290000 00007ffb`e63fe000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Ripes-v2.1.0-win-x86_64\platforms\qwindows.dll
ModLoad: 00007ffc`30940000 00007ffc`30a0d000   C:\WINDOWS\System32\OLEAUT32.dll
ModLoad: 00007ffc`2d7b0000 00007ffc`2d7c4000   C:\WINDOWS\SYSTEM32\WTSAPI32.dll
ModLoad: 00007ffc`2da30000 00007ffc`2da42000   C:\WINDOWS\SYSTEM32\kernel.appcore.dll
ModLoad: 00007ffc`2da50000 00007ffc`2e1e1000   C:\WINDOWS\SYSTEM32\windows.storage.dll
ModLoad: 00007ffc`2f380000 00007ffc`2f3ac000   C:\WINDOWS\SYSTEM32\Wldp.dll
ModLoad: 00007ffc`305a0000 00007ffc`305f5000   C:\WINDOWS\System32\shlwapi.dll
ModLoad: 00007ffc`2f850000 00007ffc`2f86f000   C:\WINDOWS\SYSTEM32\profapi.dll
ModLoad: 00007ffc`0d8d0000 00007ffc`0d8f6000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Ripes-v2.1.0-win-x86_64\styles\qwindowsvistastyle.dll
ModLoad: 00007ffc`1ed50000 00007ffc`1ed5d000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Ripes-v2.1.0-win-x86_64\imageformats\qgif.dll
ModLoad: 00007ffc`1a790000 00007ffc`1a79e000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Ripes-v2.1.0-win-x86_64\imageformats\qico.dll
ModLoad: 00007ffb`fbae0000 00007ffb`fbb4a000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Ripes-v2.1.0-win-x86_64\imageformats\qjpeg.dll
ModLoad: 00007ffc`1a380000 00007ffc`1a38c000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Ripes-v2.1.0-win-x86_64\imageformats\qsvg.dll
ModLoad: 00007ffb`fa570000 00007ffb`fa5c5000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Ripes-v2.1.0-win-x86_64\Qt5Svg.dll
ModLoad: 00007ffc`18cc0000 00007ffc`18ccf000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Ripes-v2.1.0-win-x86_64\iconengines\qsvgicon.dll
ModLoad: 00007ffc`1fb10000 00007ffc`1fd8f000   C:\WINDOWS\system32\dwrite.dll
(4130.15a0): C++ EH exception - code e06d7363 (first chance)
(4130.15a0): C++ EH exception - code e06d7363 (!!! second chance !!!)
KERNELBASE!RaiseException+0x69:
00007ffc`2fad4b59 0f1f440000      nop     dword ptr [rax+rax]

mortbopet, 2021-04-11:
Hmm... I am not quite sure about what the issue might be, and google doesn't seem to point to something obvious either. Are you running some form of anti-virus program? That might be interfering with the injected .dll.

mortbopet, 2021-04-11:
Is this issue also present if you try running the windows continuous release?
https://github.com/mortbopet/Ripes/releases/tag/continuous

trlef19, 2021-04-11:
Pre-release has also the same problem. They only one works is the 1.x.x version. I'm just using the default windows defender anti virus

mortbopet, 2021-04-11:
Alright... Let me try to get a debug build ready for you - maybe that will help us track down the issue.

trlef19, 2021-04-11:
I'll wait

mortbopet, 2021-04-12:
You can download a windows debug executable here:
https://github.com/mortbopet/Ripes/actions/runs/740362794
Try running it again, through a command line. If it still doesn't open, then try running it through WinDbg and see whether the command output changes from what we observed with the other program.

trlef19, 2021-04-12:
Hello, I ran your program through cmd and i got a runtime error and abort() was called

mortbopet, 2021-04-13:
And what is the output if you run it through WinDbg?

trlef19, 2021-04-13:
That:
0:000> g
ModLoad: 00007ffc`32160000 00007ffc`32190000   C:\WINDOWS\System32\IMM32.DLL
ModLoad: 00007ffc`31330000 00007ffc`313de000   C:\WINDOWS\System32\shcore.dll
ModLoad: 00007ffb`d5ec0000 00007ffb`d6317000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\platforms\qwindowsd.dll
ModLoad: 00007ffc`30940000 00007ffc`30a0d000   C:\WINDOWS\System32\OLEAUT32.dll
ModLoad: 00007ffc`2d7b0000 00007ffc`2d7c4000   C:\WINDOWS\SYSTEM32\WTSAPI32.dll
ModLoad: 00007ffc`2da30000 00007ffc`2da42000   C:\WINDOWS\SYSTEM32\kernel.appcore.dll
ModLoad: 00007ffc`2fd70000 00007ffc`2fdf0000   C:\WINDOWS\System32\bcryptPrimitives.dll
ModLoad: 00007ffc`2da50000 00007ffc`2e1e1000   C:\WINDOWS\SYSTEM32\windows.storage.dll
ModLoad: 00007ffc`2f380000 00007ffc`2f3ac000   C:\WINDOWS\SYSTEM32\Wldp.dll
ModLoad: 00007ffc`305a0000 00007ffc`305f5000   C:\WINDOWS\System32\shlwapi.dll
ModLoad: 00007ffc`2f850000 00007ffc`2f86f000   C:\WINDOWS\SYSTEM32\profapi.dll
ModLoad: 00007ffc`2ef00000 00007ffc`2ef4b000   C:\WINDOWS\SYSTEM32\powrprof.dll
ModLoad: 00007ffc`2ed70000 00007ffc`2ed82000   C:\WINDOWS\SYSTEM32\UMPDC.dll
ModLoad: 00007ffc`22980000 00007ffc`229e6000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\styles\qwindowsvistastyled.dll
ModLoad: 00007ffc`22d80000 00007ffc`22da0000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\imageformats\qgifd.dll
ModLoad: 00007ffc`1fd60000 00007ffc`1fd83000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\imageformats\qicnsd.dll
ModLoad: 00007ffc`226f0000 00007ffc`22710000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\imageformats\qicod.dll
ModLoad: 00007ffc`000a0000 00007ffc`00170000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\imageformats\qjpegd.dll
ModLoad: 00007ffc`1f1d0000 00007ffc`1f1eb000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\imageformats\qsvgd.dll
ModLoad: 00007ffb`fffe0000 00007ffc`0009d000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\Qt5Svgd.dll
ModLoad: 00007ffc`11730000 00007ffc`1174b000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\imageformats\qtgad.dll
ModLoad: 00007ffb`fc8d0000 00007ffb`fc99d000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\imageformats\qtiffd.dll
ModLoad: 00007ffc`11710000 00007ffc`1172a000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\imageformats\qwbmpd.dll
ModLoad: 00007ffb`f9b30000 00007ffb`f9c4c000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\imageformats\qwebpd.dll
ModLoad: 00007ffc`116e0000 00007ffc`11702000   C:\Users\ΣΤΑΥΡΟΣ Τ\Downloads\Debug_executable\Ripes-continuous-win-x86_64\iconengines\qsvgicond.dll
ModLoad: 00007ffc`0ee60000 00007ffc`0f0df000   C:\WINDOWS\system32\dwrite.dll
(3ca4.4238): C++ EH exception - code e06d7363 (first chance)
(3ca4.4238): C++ EH exception - code e06d7363 (!!! second chance !!!)
KERNELBASE!RaiseException+0x69:
00007ffc`2fad4b59 0f1f440000      nop     dword ptr [rax+rax]

mortbopet, 2021-04-13:
I see, this doesn't seem to provide any additional info as compared to the previous attempt.
I am sorry, but i am not sure about what is causing the error. My hunch is that it might be some broken or mismatched system library, which is causing the error, but it is hard to tell exactly.
If you'd like to investigate deeper, I'd suggest googling around for how to generate a crash dump of the application. Then we might be able to figure out something new. 

If you haven't already, you can run Ripes through a linux system, possibly in a virtual machine. You can download [virtualbox](https://www.virtualbox.org/) alongside a [pre-installed image of i.e., Ubuntu](https://www.osboxes.org/ubuntu/) which will have you up and running in no time.

trlef19, 2021-04-13:
I'll check it out and give you some feedback

trlef19, 2021-04-14:
Hello, I managed to create crash dump and here is the output:
User Mini Dump File: Only registers, stack and portions of memory are available

************* Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       srv*
Symbol search path is: srv*
Executable search path is: 
Windows 10 Version 19042 MP (8 procs) Free x64
Product: WinNt, suite: SingleUserTS Personal
Edition build lab: 19041.1.amd64fre.vb_release.191206-1406
Machine Name:
Debug session time: Thu Apr 15 00:54:28.000 2021 (UTC + 3:00)
System Uptime: not available
Process Uptime: 0 days 0:00:04.000
......................................................
This dump file has an exception of interest stored in it.
The stored exception information can be accessed via .ecxr.
(e20.17d0): Security check failure or stack buffer overrun - code c0000409 (first/second chance not available)
Subcode: 0x7 FAST_FAIL_FATAL_APP_EXIT 
For analysis of this file, run !analyze -v
ucrtbase!abort+0x4e:
00007ffc`7409286e cd29            int     29h

mortbopet, 2021-04-16:
Great - next we need to analyse the cause of the exception. I must admit i do not have that much experience in analysing crashes on Windows, so your best bet is probably also to google a bit here. I know that the dump file can be analyzed by windbg (either old or new version), and maybe running the `!analyze -v` command will show something interesting.

trlef19, 2021-04-16:
I did it. Here is the output:
0000004c`dc2ff8e0 00007ff6`ba6efe5f     : 00000207`00000001 00000207`257f5b50 00000207`00000005 0000004c`dc2ff990 : Ripes+0xa149b
0000004c`dc2ff980 00007ff6`ba69fdc4     : 00000207`281149a0 00000207`2780d4c0 00000207`298d12a0 00000207`00000000 : Ripes+0x9fe5f
0000004c`dc2ffa40 00007ff6`ba69dcd5     : 0000004c`dc2ffb40 00000207`2574b250 00000207`281149a0 00000207`2580c4f0 : Ripes+0x4fdc4
0000004c`dc2ffb00 00007ff6`ba775d27     : 00000000`00000001 00000000`00000000 00000207`257e5f40 00000207`25746010 : Ripes+0x4dcd5
0000004c`dc2ffbe0 00007ff6`ba774d4e     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : Ripes+0x125d27
0000004c`dc2ffc70 00007ffc`74da7034     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : Ripes+0x124d4e
0000004c`dc2ffcb0 00007ffc`766e2651     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : kernel32!BaseThreadInitThunk+0x14
0000004c`dc2ffce0 00000000`00000000     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!RtlUserThreadStart+0x21

SYMBOL_NAME:  ucrtbase!abort+4e

MODULE_NAME: ucrtbase

IMAGE_NAME:  ucrtbase.dll

STACK_COMMAND:  ~0s ; .ecxr ; kb

FAILURE_BUCKET_ID:  FAIL_FAST_FATAL_APP_EXIT_c0000409_ucrtbase.dll!abort

OS_VERSION:  10.0.19041.1

BUILDLAB_STR:  vb_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

IMAGE_VERSION:  10.0.19041.789

FAILURE_ID_HASH:  {e31753ac-c98a-8055-3663-47e707543d20}

Followup:     MachineOwner
---------

mortbopet, 2021-07-09:
Closed due to being unable to replicate, most likely machine-specific.

closed 2021-07-09 by mortbopet