mortbopet/Ripes #84: Windows version does not zero-terminate strings

Cirromulus, 2021-04-26:
Hi!
I have the issue that, under windows 10, strings with `length % 4 != 0` are not null-terminated.
This leads to printing two adjacent strings.
This does not happen on my linux machine.

The following code produces two different binaries:

```riscv-asm
.data
n: .word 25		# Hier Wert für n eintragen
root: .word 5		# Hier das erwartete Ergebnis eintragen

strCalc: .string "Berechne Quadratwurzel von "
strErg: .string ". Das Ergebnis "
strIs: .string " ist "
strKorr: .string "korrekt."	# If len % 4 == 0, no zero byte is added
strFals: .string "falsch."
```
Windows version:

[image]

Linux (or on windows with extra space-character):
[image]

I am using v2.1.0 on both machines.

mortbopet, 2021-04-26:
hi @Cirromulus, thank you for raising the issue. It should be fixed in 2aa04f7 and the fix is available in the [most recent continuous release](https://github.com/mortbopet/Ripes/releases/tag/continuous), if you'd like to test/verify.

Cirromulus, 2021-04-26:
It works with the continuus release version.
Thanks!

closed 2021-04-26 by Cirromulus