mortbopet/Ripes #85: Error Qt5Core

pzck3663, 2021-05-21:
Hye , im sorry , why im having problem " find_package(Qt5Core REQUIRED) " , how to solve it ?

mortbopet, 2021-05-21:
Hello @pzck3663, make sure that your Qt installation is available in your CMake prefix path, see https://doc.qt.io/qt-5/cmake-get-started.html. Also, try googling around on how to get a "hello world" Qt+CMake program to build - there is plenty of documentation out there!

pzck3663, 2021-05-21:
So i need to install the Qt Software ?

pzck3663, 2021-05-21:
So i need to build Qt+Cmake using Qt Software ? sorry im new to this hmm. But the output said i dont have this file :

Error CMake Error at D:\users\Dekstop\CAO Software\PROJECT\CMakeLists.txt:31 (find_package):
 By not providing "FindQt5Core.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5Core", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5Core" with any
  of the following names:

    Qt5CoreConfig.cmake
    qt5core-config.cmake

  Add the installation prefix of "Qt5Core" to CMAKE_PREFIX_PATH or set
  "Qt5Core_DIR" to a directory containing one of the above files.  If
  "Qt5Core" provides a separate development package or SDK, be sure it has
  been installed.	Ripes	D:\users\Dekstop\CAO Software\PROJECT\CMakeLists.txt	31

mortbopet, 2021-05-21:
As a minimum, you need to have a Qt installation available. 
Try checking out this post: https://stackoverflow.com/questions/61433498/cmake-prefix-path-for-qt5

It is the environment variable `CMAKE_PREFIX_PATH` which you need to modify (or add), to point to the path of your Qt installation.

pzck3663, 2021-05-21:
What component that should i choose in installing Qt ? Is there anything that i should do before modify or aadd to point to the path of my Qt Installation ?

pzck3663, 2021-05-21:
Did i need to tick specific component in Qt during installation ?

mortbopet, 2021-05-21:
Apart from the "regular" Qt installation, you also need to install the "Qt Charts" component. You can always do this afterwards through the Qt Maintenace Tool (if you've the Qt installer, this program is available the same place where QtCreator is located).

pzck3663, 2021-05-21:
Owh okay , ill do using Qt Maintenance tool because i install the Qt 6 something vers because theres only option with Qt5.15 Qt6 somthing and Custom . Thank you , ill try it now

pzck3663, 2021-05-21:
But when i try to make new project or use the CMakelist.txt file , Qt dont show any suitable kits . Is this can be download in Qt Maintenance and what component should i download ?

mortbopet, 2021-05-21:
I would recommend you to search online for how to get a Qt environment up and running on Windows. There are 100's of StackOverflow questions as well as various Qt-provided tutorials for doing this.

pzck3663, 2021-05-21:
okay

pzck3663, 2021-05-22:
i already try to put the directory of my Qt5Core but it said error , the  CMAKE_PREFIX_PATH should put into CMakelist.txt right ?
and my Qt installation file is at another disk.

mortbopet, 2021-05-22:
Again, i would recommend you to look online on stack overflow and Qt tutorials, and make sure that you can build an example Qt program, i.e. https://github.com/jasondegraw/Qt-CMake-HelloWorld.
If that works, and there is still an issue with building Ripes, then please let me know.

mortbopet, 2021-06-02:
Leaving as closed due to (most likely) not being related to Ripes itself.

closed 2021-06-02 by mortbopet