mortbopet/Ripes #88: Error about "Tests/Ripes/src/version/version.cmake: No such file or directory"

Petross404, 2021-05-22:
I cloned recursively the repository and when I tried to build it with make I got the following:
```
[  1%] Generating version/dummy.cpp
/bin/sh: line 1: Tests/Ripes/src/version/version.cmake: No such file or directory
/bin/sh: line 1: Tests/Ripes: No such file or directory
make[2]: *** [src/CMakeFiles/version_file.dir/build.make:73: src/version/dummy.cpp] Error 127
make[1]: *** [CMakeFiles/Makefile2:725: src/CMakeFiles/version_file.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "/home/petros/projects/Ebuilds&Tests/Ripes/build/src" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```

How to debug this?

mortbopet, 2021-05-22:
Looking at the discrepancy between the source directory `/home/petros/projects/Ebuilds&Tests/Ripes/build/src` and the paths printed earlier (i.e., `Tests/Ripes/src/version/version.cmake`, it might be that the `&` character in the path is messing things up.

Petross404, 2021-05-24:
> Looking at the discrepancy between the source directory `/home/petros/projects/Ebuilds&Tests/Ripes/build/src` and the paths printed earlier (i.e., `Tests/Ripes/src/version/version.cmake`, it might be that the `&` character in the path is messing things up.

Interesting, I will take a look at this. Thank you for your time.

Petross404, 2021-06-06:
> Looking at the discrepancy between the source directory `/home/petros/projects/Ebuilds&Tests/Ripes/build/src` and the paths printed earlier (i.e., `Tests/Ripes/src/version/version.cmake`, it might be that the `&` character in the path is messing things up.

Yes, the & character was the culprit here, thank you.

closed 2021-06-06 by Petross404