mortbopet/Ripes #89: Error when using the "mingw32-make" or "jom"

JuanJRom, 2021-05-30:
Hello good day, 

I have problems when using the "mingw32-make" or "jom" it throws me the following error

[ 66%] Building CXX object src/CMakeFiles/ripes_lib.dir/processorhandler.cpp.obj
[ 67%] Linking CXX executable tst_leros.exe
[ 67%] Built target tst_leros
C:/Qt/Tools/mingw810_64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/bin/as.exe: CMakeFiles\ripes_lib.dir\processorhandler.cpp.obj: too many sections (40575)
C:\Users\JUANRO~1\AppData\Local\Temp\ccWebnW4.s: Assembler messages:
C:\Users\JUANRO~1\AppData\Local\Temp\ccWebnW4.s: Fatal error: can't write 92 bytes to section .text of CMakeFiles\ripes_lib.dir\processorhandler.cpp.obj: 'File too big'
C:/Qt/Tools/mingw810_64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/bin/as.exe: CMakeFiles\ripes_lib.dir\processorhandler.cpp.obj: too many sections (40575)
C:\Users\JUANRO~1\AppData\Local\Temp\ccWebnW4.s: Fatal error: can't close CMakeFiles\ripes_lib.dir\processorhandler.cpp.obj: File too big
jom: C:\Users\Juan Romero\Documents\Ripes\build\src\CMakeFiles\ripes_lib.dir\build.make [src\CMakeFiles\ripes_lib.dir\processorhandler.cpp.obj] Error 1
jom: C:\Users\Juan Romero\Documents\Ripes\build\CMakeFiles\Makefile2 [src\CMakeFiles\ripes_lib.dir\all] Error 2
jom: C:\Users\Juan Romero\Documents\Ripes\build\Makefile [all] Error 2

Do you know what is causing this error and how can I fix it?

Thank you

mortbopet, 2021-05-30:
Hi @JuanJRom 
Did you try googling the error? There seems to be a few common fixes to the "too many sections" issue:
https://github.com/assimp/assimp/issues/2067
https://digitalkarabela.com/mingw-w64-how-to-fix-file-too-big-too-many-sections/

JuanJRom, 2021-05-31:
Hi @mortbopet

I tried googling the error, I found adding "-mbig-obj" when calling cmake, but it didn't work for me, then with the information it gave me, I opened "CMakeLists.txt" and added "CMAKE_CXX_FLAGS = - Wa, - mbig-obj "and this fixed the error, , the simulator is working now.

Thanks!

closed 2021-05-31 by JuanJRom