mortbopet/Ripes #91: Error in defining strings in .data section

MassimilianoMancini, 2021-06-03:
In version 2.2.0 (win64) defining following strings in .data section throws compilation errors as reported in remarks:
```
.data

# Unmatched parenthesis
anystring1:   .string   "foo("
anystring2:   .string   "foo)"

# Stray '.' in line
anystring3:   .string   "foo(.)"
```

mortbopet, 2021-06-03:
Thank you for finding this - this seems to be issues with the expression evaluator.

mortbopet, 2021-06-03:
@MassimilianoMancini when you have time, please check whether the issue has been fixed by ae930d1.
The fix is available in the latest continuous build: https://github.com/mortbopet/Ripes/releases/tag/continuous

MassimilianoMancini, 2021-06-03:
@mortbopet Checked and I confirm that it has been fixed. Thank you

closed 2021-06-03 by mortbopet