mortbopet/Ripes #93: Ripes can't work with riscv64-unknown-linux-gnu-gcc on Gentoo Linux.

Petross404, 2021-06-08:
Hi,

I am testing Ripes mostly to learn about RISC-V assembly, but I have trouble setting the compiler up. I am running Gentoo and I am using crossdev to create cross-compilers. The tuple of the resulting compiler I created is named "riscv64-unknown-linux-gnu-" but "riscv64-unknown-elf-" didn't work either.

What tuple should I use for crossdev?

mortbopet, 2021-06-08:
Hi @Petross404 
I am unfamiliar with crossdev, so i cannot give advice on specifics wrt. that.
The way Ripes tests a compiler is to try and compile a test program with flags based on the ISA of the current processor model,  using whatever executable that you selected in the compiler path setting.

I've just pushed a change which also shows the compile command if running the compiler failed - this is available if you download the latest continuous release.
If you can run this command in a console outside of Ripes, and have it run successfully, then the compiler _should_ work inside of Ripes - in any case, make sure that you can successfully compile a program with you compiler, outside of Ripes.
No matter what, the `riscv64-unknown-elf` should be the one to use, since this is the bare-metal toolchain.

Petross404, 2021-06-08:
> Hi @Petross404
> I am unfamiliar with crossdev, so i cannot give advice on specifics wrt. that.
> The way Ripes tests a compiler is to try and compile a test program with flags based on the ISA of the [...]

I haven't tested your fix yet, changing the ld flags to "-static -nostdlib" fixed it for me with /usr/bin/riscv64-unknown-elf-gcc. Thank you for your help.

closed 2021-06-08 by Petross404

mortbopet, 2021-06-08:
great! `static-libgcc` or `-static` should ideally be present as a default argument in the linker arguments... Maybe it should be made impossible to remove this argument from the linker arguments, seeing as non-statically linked executables won't be able to run anyways...

Petross404, 2021-06-11:
I think they should be editable but under an "advanced" option with a big, fat warning. I mean, you never know. But anyway, congratulations for your effort and thanks for your support.