mortbopet/Ripes #98: Crashes upon launch on MacOS 11.3.1

kurmasz, 2021-06-17:
The program crashes upon launch (before any windows open) on 2020 MacBook pro running MacOS 11.3.1

macOS problem report below.

If it helps, after downloading the file, I had to run "chmod 755 -R" before it would launch.  Perhaps there is a file / directory that needs different permissions?

Process:               Ripes [46654]
Path:                  /Applications/Ripes-2.2.1-mac-x86_64.app/Contents/MacOS/Ripes
Identifier:            ???
Version:               ??? (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Ripes [46654]
User ID:               47804697

Date/Time:             2021-06-17 11:53:16.882 -0400
OS Version:            macOS 11.3.1 (20E241)
Report Version:        12
Bridge OS Version:     5.3 (18P4556)
Anonymous UUID:        1E3F4501-437C-8A73-1844-C79AA3616642

Sleep/Wake UUID:       9A878C9F-5D33-4990-A998-51CF396EBF9B

Time Awake Since Boot: 1500000 seconds
Time Since Wake:       11000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000028
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [46654]

VM Regions Near 0x28:
--> 
    __TEXT                      106980000-106f58000    [ 5984K] r-x/r-x SM=COW  /Applications/Ripes-2.2.1-mac-x86_64.app/Contents/MacOS/Ripes

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0                                 	0x0000000106a43725 Ripes::CacheTabWidget::CacheTabWidget(QWidget*) + 709
1                                 	0x0000000106a3d7b1 0x106980000 + 776113
2                                 	0x0000000106a3d1d5 Ripes::CacheTab::CacheTab(QToolBar*, QWidget*) + 101
3                                 	0x0000000106af01c0 Ripes::MainWindow::MainWindow(QWidget*) + 1520
4                                 	0x0000000106985c79 main + 73
5   libdyld.dylib                 	0x00007fff20634f3d start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff20615484 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff20615484 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff20615484 start_wqthread + 0

Thread 4:
0   libsystem_pthread.dylib       	0x00007fff20615484 start_wqthread + 0

Thread 5:: com.apple.CFSocket.private
0   libsystem_kernel.dylib        	0x00007fff205ec66e __select + 10
1   com.apple.CoreFoundation      	0x00007fff207391e3 __CFSocketManager + 630
2   libsystem_pthread.dylib       	0x00007fff20619954 _pthread_start + 224
3   libsystem_pthread.dylib       	0x00007fff206154a7 thread_start + 15

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000106a44050  rcx: 0x0000000000000000  rdx: 0x0000000000000001
[... 400 lines cut ...]
Memory Module: BANK 2/ChannelB-DIMM0, 16 GB, DDR4, 2667 MHz, Micron, -
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Feb 16 2021 03:05:58 version 9.30.444.10.32.5.67 FWID 01-3d719d60
Bluetooth: Version 8.0.4d18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Network Service: iPhone, Ethernet, en9
USB Device: My Passport 260F
USB Device: Hub
USB Device: USB 3.1 Bus
USB Device: Hub
USB Device: iPhone
USB Device: Hub in Apple Pro Keyboard
USB Device: Apple Pro Keyboard
USB Device: HL-2140 series
USB Device: Composite Device
USB Device: Apple T2 Bus
USB Device: Composite Device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5

mortbopet, 2021-06-18:
Hi @kurmasz, thank you for reporting the error. I unfortunately don't have an OSX machine so the testing that i am able to do is very limited, hence things like this can happen.
Regarding the permission thing, i do notice that as well, and I'll try to figure out why the executable permission is getting lost somewhere.

Reading the error, it seems to be somewhere in the constructor of [CacheTabWidget](https://github.com/mortbopet/Ripes/blob/master/src/cachetabwidget.cpp), however, it is hard to tell beyond that.
I see (over on the discussions page) that you are trying to get Ripes building locally - Great! If you are able to get that sorted out, and run in debug mode, then we'll be able to get a better view at the exact location where this is happening.

kurmasz, 2021-06-18:
I got it compiled.  It crashed before opening any window, but the line numbers in the dump were slightly different.

How do I run it in debug mode?  (I've never done macOS development.)

mortbopet, 2021-06-18:
The permissions issue should be fixed in 06b0509.
Which editor/IDE are you using? If you do not have prior experience in debugging C++ applications, i can recommend using QtCreator (an IDE that comes along with a Qt download, if you downloaded Qt through https://www.qt.io/download). If doing so, you can then refer to this page for information on debugging: https://doc.qt.io/qtcreator/creator-debugging.html

kurmasz, 2021-06-18:
Since there is a CMake file, I wonder if CLion would be a good choice?

mortbopet, 2021-06-19:
CLion would work as well, it just takes a little bit more effort to ensure that the Qt environment is set up properly. The most important thing is to properly make the `CMAKE_PREFIX_PATH` available in either your global `PATH` variable, or specifically for the CMake project (https://www.jetbrains.com/help/clion/qt-tutorial.html#cmake-settings).

kurmasz, 2021-06-21:
I got it to launch in debug mode.  Here is the stack trace:

QWidget::size() const qwidget.h:866
Ripes::CacheTabWidget::CacheTabWidget(QWidget*) cachetabwidget.cpp:50
Ripes::CacheTabWidget::CacheTabWidget(QWidget*) cachetabwidget.cpp:31
Ripes::Ui_CacheTab::setupUi(QWidget*) ui_cachetab.h:55
Ripes::CacheTab::CacheTab(QToolBar*, QWidget*) cachetab.cpp:12
Ripes::CacheTab::CacheTab(QToolBar*, QWidget*) cachetab.cpp:11
Ripes::MainWindow::MainWindow(QWidget*) mainwindow.cpp:70
Ripes::MainWindow::MainWindow(QWidget*) mainwindow.cpp:36
main main.cpp:19
start 0x00007fff20634f3d
start 0x00007fff20634f3d

qwidget.h:866 is the return statement below

inline QSize QWidget::size() const
{ return data->crect.size(); }

So, my guess is that data is still null.   (However, QWidget is not part of the project, so I can't mess with it.)

mortbopet, 2021-06-21:
Looks like there is a possible nullptr exception at cachetabwidget.cpp:50.
What if you substitute that line with:
```
    // Store default close button size for future use
    if (auto* tabButton = m_ui->tabWidget->tabBar()->tabButton(0, QTabBar::RightSide)) {
        m_defaultTabButtonSize = tabButton->size();
    }
```

kurmasz, 2021-06-21:
When I did that, another nullptr came up on line 53.  However,  the change below allowed the app to launch:

    if (auto* tabButton = m_ui->tabWidget->tabBar()->tabButton(0, QTabBar::RightSide)) {
        m_defaultTabButtonSize = tabButton->size();
        // Disable closing for L1 caches and add tab
        m_ui->tabWidget->tabBar()->tabButton(0, QTabBar::RightSide)->resize(0, 0);
        m_ui->tabWidget->tabBar()->tabButton(1, QTabBar::RightSide)->resize(0, 0);
    }

I'll have to play around with it later and see if any other bugs come up.

kurmasz, 2021-06-21:
For future reference:  When I created the CLion project over top of the cloned repo, CLion replaced CMakeLists.txt and main.cpp with it's own default files.  I had to delete those and check out the versions from the repo.  Once I did that, everything worked as expected (no need to manually set the CMAKE_PREFIX_PATH.

mortbopet, 2021-06-21:
> When I did that, another nullptr came up on line 53. However, the change below allowed the app to launch:
>
> ```
> if (auto* tabButton = m_ui->tabWidget->tabBar()->tabButton(0, QTabBar::RightSide)) [...]

Woopsie, didn't spot the pretty much identical error below - sorry!

kurmasz, 2021-06-21:
Is that an error, or a bug in the Mac implementation?  If it is an error, why doesn't it affect the Linux version?  (Just curious)

mortbopet, 2021-06-21:
My hunch is that, on Linux and Windows, a TabBar has some default buttons present, whereas this is not the case for Mac. Hence, when calling https://doc.qt.io/qt-5/qtabbar.html#tabButton, a nullptr is returned (which it also warns about).

mortbopet, 2021-07-09:
@kurmasz can you check top of master/the latest continuous release to check if this is still an issue? If not, please close the issue 👍 .

kurmasz, 2021-07-09:
Will do, but it might be a couple days.

Zachary Kurmas
Associate Professor of Computer Science
Grand Valley State University
http://www.cis.gvsu.edu/~kurmasz

kurmasz, 2021-07-10:
I cloned a fresh copy of the repo, built it, and verified that it launched cleanly on macOS.  

(It crashed when I went to quit; but, I'll open another issue later 😃 )

closed 2021-07-10 by kurmasz

mortbopet, 2021-07-10:
Thank you for letting me know - i was about to prepare a new release, but given that, i'll hold off a bit for now...