mortbopet/Ripes #102: Memory viewer isn't displaying anything

deananorth, 2021-07-14:
Operating system: Windows 10 21H1
Version: Continuous

Expected result: Memory display should show values in memory.
Actual result: Memory display is blank. Combo boxes at the bottom of the screen do not seem to work. No error messages appear as to why memory.

Ripes Continuous
[image]

Ripes 2.2.1
[image]

mortbopet, 2021-07-14:
Hi @deananorth, this should hopefully have been fixed in 5d8ee3c7dc9b20968f51557e95ea4c1146d02b44 - please verify in the latest continuous release - thank you! :+1:

deananorth, 2021-07-14:
[image]

Tried the latest continuous released (Commit ce8d568, released 10 hours ago as of this post)
it seems to have worked.

Thanks for your help.

closed 2021-07-14 by deananorth