mortbopet/Ripes #103: Crashes upon Exit in macOS

kurmasz, 2021-07-15:
Ripes crashes when you exit using Ripes -> Quit Ripes

I ran it in debug mode using CLion.  It crashes in the Ripes::CacheInterface destructor at the "ud2" instruction. This is what I see (but, I don't quite understand how Ripes::CacheInterface is implemented).

Ripes::CacheInterface::~CacheInterface():
	pushq  %rbp
	movq   %rsp, %rbp
	movq   %rdi, -0x8(%rbp)
	ud2
	nopw   (%rax,%rax)

It looks like the macro may be generating invalid code.

--------

If it helps, this is the dump that macOS produces:

Process:               Ripes [51847]
Path:                  /Users/USER/Documents/*/Ripes.app/Contents/MacOS/Ripes
Identifier:            ???
Version:               ??? (???)
Code Type:             X86-64 (Native)
Parent Process:        clion [49279]
Responsible:           clion [49279]
User ID:               47804697

Date/Time:             2021-07-15 15:37:37.321 -0400
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Bridge OS Version:     5.4 (18P4663)
Anonymous UUID:        1E3F4501-437C-8A73-1844-C79AA3616642

Sleep/Wake UUID:       EA9FB1E2-6DC4-421A-AA03-486F45F6C717

Time Awake Since Boot: 430000 seconds
Time Since Wake:       950 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [51847]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0                                 	0x00000001084f9408 Ripes::CacheInterface::~CacheInterface() + 8
1                                 	0x00000001089014fb std::__1::default_delete<Ripes::ProcInfoBase>::operator()(Ripes::ProcInfoBase*) const + 43
2                                 	0x000000010890147f std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> >::reset(Ripes::ProcInfoBase*) + 95
3                                 	0x0000000108901419 std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> >::~unique_ptr() + 25
4                                 	0x00000001089013f5 std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> >::~unique_ptr() + 21
5                                 	0x00000001089013ce std::__1::pair<Ripes::ProcessorID const, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >::~pair() + 30
6                                 	0x00000001089013a5 std::__1::pair<Ripes::ProcessorID const, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >::~pair() + 21
7                                 	0x0000000108901389 void std::__1::allocator_traits<std::__1::allocator<std::__1::__tree_node<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, void*> > >::__destroy<std::__1::pair<Ripes::ProcessorID const, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > > >(std::__1::integral_constant<bool, false>, std::__1::allocator<std::__1::__tree_node<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, void*> >&, std::__1::pair<Ripes::ProcessorID const, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >*) + 25
8                                 	0x00000001089012cd void std::__1::allocator_traits<std::__1::allocator<std::__1::__tree_node<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, void*> > >::destroy<std::__1::pair<Ripes::ProcessorID const, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > > >(std::__1::allocator<std::__1::__tree_node<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, void*> >&, std::__1::pair<Ripes::ProcessorID const, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >*) + 29
9                                 	0x000000010890122f std::__1::__tree<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, std::__1::__map_value_compare<Ripes::ProcessorID, std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, std::__1::less<Ripes::ProcessorID>, true>, std::__1::allocator<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > > > >::destroy(std::__1::__tree_node<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, void*>*) + 127
10                                	0x00000001089011e3 std::__1::__tree<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, std::__1::__map_value_compare<Ripes::ProcessorID, std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, std::__1::less<Ripes::ProcessorID>, true>, std::__1::allocator<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > > > >::destroy(std::__1::__tree_node<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, void*>*) + 51
11                                	0x00000001089011e3 std::__1::__tree<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, std::__1::__map_value_compare<Ripes::ProcessorID, std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, std::__1::less<Ripes::ProcessorID>, true>, std::__1::allocator<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > > > >::destroy(std::__1::__tree_node<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, void*>*) + 51
12                                	0x00000001089011a8 std::__1::__tree<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, std::__1::__map_value_compare<Ripes::ProcessorID, std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, std::__1::less<Ripes::ProcessorID>, true>, std::__1::allocator<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > > > >::~__tree() + 40
13                                	0x0000000108901175 std::__1::__tree<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, std::__1::__map_value_compare<Ripes::ProcessorID, std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > >, std::__1::less<Ripes::ProcessorID>, true>, std::__1::allocator<std::__1::__value_type<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > > > >::~__tree() + 21
14                                	0x0000000108901155 std::__1::map<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> >, std::__1::less<Ripes::ProcessorID>, std::__1::allocator<std::__1::pair<Ripes::ProcessorID const, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> > > > >::~map() + 21
15                                	0x0000000108901135 std::__1::map<Ripes::ProcessorID, std::__1::unique_ptr<Ripes::ProcInfoBase, std::__1::default_delete<Ripes::ProcInfoBase> >, std::__1::less<Ripes::ProcessorID>, std::__1::allocator [... cut ...]
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: Apple T2 Bus
USB Device: Composite Device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5

mortbopet, 2021-07-16:
Thank you for reporting the issue - the jump between
```
0 0x00000001084f9408 Ripes::CacheInterface::~CacheInterface() + 8
1 0x00000001089014fb std::__1::default_deleteRipes::ProcInfoBase::operator()(Ripes::ProcInfoBase*) const + 43
```
does seem a bit strange. I've pushed a few memory leak fixes + added a virtual destructor to CacheInterface (which should have been there from the beginning).
Do you mind checking again, building from the head of master? And if the problem persists, maybe also try adding `-fsanitize=address` to `CMAKE_CXX_FLAGS` ( https://github.com/mortbopet/Ripes/blob/master/CMakeLists.txt#L16).

kurmasz, 2021-07-20:
When rebuilding the current head of master, I got the same crash.  

But, after adding a destructor to ProcInfoBase, it seems to work.

(Specifically, I added virtual ~ProcInfoBase() {} as line 62 of processorregistry.h.)

If it matters, this is the remaining warning I have when building:

[100%] Linking CXX static library librv5s_no_fw_hz.a
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: librv5s_no_fw_hz.a(mocs_compilation.cpp.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: librv5s_no_fw_hz.a(mocs_compilation.cpp.o) has no symbols
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: archive library: librv5s_no_fw_hz.a the table of contents is empty (no object file members in the library define global symbols)
[100%] Built target rv5s_no_fw_hz

mortbopet, 2021-07-21:
Great! - seems like there might be a few of those missing destructors. I'll go ahead and push your fix.

Regarding the no symbols error; this is probably an artifact of how each processor is added to the CMake project.
In https://github.com/mortbopet/Ripes/blob/master/src/processors/CMakeLists.txt, `create_vsrtl_processor` will create a static library for each processor. However, most (all?) of the processor descriptions contain purely `.h` files, so i suspect that this is the reason for the error - nothing is actually being compiled into these libraries, and so they contain no symbols.

kurmasz, 2021-07-21:
That makes sense.

I didn't see any other "missing destructor" warnings.

mortbopet, 2021-07-21:
Just for a final verification, could you briefly check the continuous release to verify the issue has been fixed?

kurmasz, 2021-07-21:
Yes it worked.  

Specifically, when I did a clean build of the head, I was able to launch and quit the program with no crashes.

closed 2021-07-21 by kurmasz