mortbopet/Ripes #104: Unable to run SPKIE on Apple M1 chip

hossein1387, 2021-07-17:
I am using Ripe release version `Ripes-2.2.1-mac-x86_64.app` and hoping to run Ripe through OSX rosetta but it crashed with the following problem report, any plan to support Macosx with Arm M1 chip?

```

Process:               Ripes [33314]
Path:                  /private/var/folders/*/Ripes-2.2.1-mac-x86_64.app/Contents/MacOS/Ripes
Identifier:            ???
Version:               ??? (???)
Code Type:             X86-64 (Translated)
Parent Process:        ??? [1]
Responsible:           Ripes [33314]
User ID:               503

Date/Time:             2021-07-17 18:05:00.350 -0400
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Anonymous UUID:        7C139750-03EF-B06F-E173-73B6C4A55F4C

Sleep/Wake UUID:       F51CBABC-B701-4043-904C-F53749F3F04C

Time Awake Since Boot: 77000 seconds
Time Since Wake:       6800 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000028
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [33314]

VM Regions Near 0x28:
--> 
    __TEXT                      100472000-100a4a000    [ 5984K] r-x/r-x SM=COW  /var/folders/*/Ripes-2.2.1-mac-x86_64.app/Contents/MacOS/Ripes

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0                                 	0x0000000100535725 Ripes::CacheTabWidget::CacheTabWidget(QWidget*) + 709
1                                 	0x000000010052f7b1 0x100472000 + 776113
2                                 	0x000000010052f1d5 Ripes::CacheTab::CacheTab(QToolBar*, QWidget*) + 101
3                                 	0x00000001005e21c0 Ripes::MainWindow::MainWindow(QWidget*) + 1520
4                                 	0x0000000100477c79 main + 73
5   libdyld.dylib                 	0x00007fff2033df5d start + 1

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	0x00007ffdffd1d858 0x7ffdffd1c000 + 6232
1   runtime                       	0x00007ffdffd27d50 0x7ffdffd1c000 + 48464
2   runtime                       	0x00007ffdffd28a20 0x7ffdffd1c000 + 51744

Thread 2:
0   runtime                       	0x00007ffdffd7699c 0x7ffdffd1c000 + 371100

Thread 3:
0   runtime                       	0x00007ffdffd7699c 0x7ffdffd1c000 + 371100

Thread 4:
0   runtime                       	0x00007ffdffd7699c 0x7ffdffd1c000 + 371100

Thread 5:
0   runtime                       	0x00007ffdffd7699c 0x7ffdffd1c000 + 371100

Thread 6:
0   runtime                       	0x00007ffdffd7699c 0x7ffdffd1c000 + 371100

Thread 7:: com.apple.CFSocket.private
[... 387 lines cut ...]
__OBJC_RW                         2480K        2 
__TEXT                           230.6M      324 
__UNICODE                          588K        1 
mapped file                        4.5G      413 
shared memory                      788K       12 
unshared pmap                     5808K        3 
===========                     =======  ======= 
TOTAL                              6.3G     1768 
TOTAL, minus reserved VM space     5.7G     1768 

Model: MacBookPro17,1, BootROM 6723.120.36, proc 8:4:4 processors, 16 GB, SMC 
Graphics: kHW_AppleM1Item, Apple M1, spdisplays_builtin
Memory Module: LPDDR4
AirPort: spairport_wireless_card_type_airport_extreme, wl0: Apr 28 2021 19:53:33 version 18.50.40.7.7.8.119 FWID 01-e6894c69
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: USB3.0 Hub
USB Device: AX88179
USB Device: USB2.0 Hub
USB Device: USB 2.0 BILLBOARD
USB Device: USB 3.1 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

mortbopet, 2021-07-18:
This looks to be related to #98, so I'd suggest trying the [latest continuous release](https://github.com/mortbopet/Ripes/releases/tag/continuous), to see if the problem persists.

In general, supporting mac systems (pre- and post M1) is a bit tricky, since I don't own a mac machine, and haven't been able to get a VM working reliably... So for now, I have to rely on other people trying the app on their macs, and submitting PRs when things don't work.

hossein1387, 2021-07-18:
Thanks @mortbopet, I am able to run Ripes with the latest continuous release. I was unable to make Ripe recognize my compiler. This issue https://github.com/mortbopet/Ripes/issues/101 seems to be my exact same problem. One last thing, as soon as I close Ripes, I get: `Ripes quit Unexpectedly` with the following message:

```
Process:               Ripes [99962]
Path:                  /Volumes/VOLUME/*/Ripes-continuous-mac-x86_64.app/Contents/MacOS/Ripes
Identifier:            ???
Version:               ??? (???)
Code Type:             X86-64 (Translated)
Parent Process:        ??? [1]
Responsible:           Ripes [99962]
User ID:               503

Date/Time:             2021-07-18 19:22:35.065 -0400
OS Version:            macOS 11.4 (20F71)
Report Version:        12

Time Awake Since Boot: 100000 seconds
Time Since Wake:       1400 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [99962]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0                                 	0x000000010256d176 0x102414000 + 1413494
1                                 	0x000000010256d14a 0x102414000 + 1413450
2                                 	0x000000010256d14a 0x102414000 + 1413450
3   libsystem_c.dylib             	0x00007fff20251d35 __cxa_finalize_ranges + 316
4   libsystem_c.dylib             	0x00007fff20252020 exit + 53
5   libdyld.dylib                 	0x00007fff2033df64 start + 8

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	0x00007ffdffcef858 0x7ffdffcee000 + 6232
1   runtime                       	0x00007ffdffcf9d50 0x7ffdffcee000 + 48464
2   runtime                       	0x00007ffdffcfaa20 0x7ffdffcee000 + 51744

Thread 2:
0   runtime                       	0x00007ffdffd4899c 0x7ffdffcee000 + 371100

Thread 3:
0   runtime                       	0x00007ffdffd4899c 0x7ffdffcee000 + 371100

Thread 4:
0   runtime                       	0x00007ffdffd4899c 0x7ffdffcee000 + 371100

Thread 5:: com.apple.CFSocket.private
0   ???                           	0x00007ffe942deab8 ???
1   libsystem_kernel.dylib        	0x00007fff202f5656 __select + 10
2   libsystem_pthread.dylib       	0x00007fff203228fc _pthread_start + 224
3   libsystem_pthread.dylib       	0x00007fff2031e443 thread_start + 15

Thread 6:
0   runtime                       	0x00007ffdffd4899c 0x7ffdffcee000 + 371100

Thread 7:
0   runtime                       	0x00007ffdffd4899c 0x7ffdffcee000 + 371100

Thread 8:: com.apple.NSEventThread
0   ???                           	0x00007ffe942deab8 ???
1   libsystem_kernel.dylib        	0x00007fff202ed2ba mach_msg_trap + 10
[... 403 lines cut ...]
__TEXT                           239.0M      329 
__UNICODE                          588K        1 
mapped file                        4.5G      432 
shared memory                      800K       15 
unshared pmap                     8736K        5 
===========                     =======  ======= 
TOTAL                              6.8G     1941 
TOTAL, minus reserved VM space     5.8G     1941 

Model: MacBookPro17,1, BootROM 6723.120.36, proc 8:4:4 processors, 16 GB, SMC 
Graphics: kHW_AppleM1Item, Apple M1, spdisplays_builtin
Memory Module: LPDDR4
AirPort: spairport_wireless_card_type_airport_extreme, wl0: Apr 28 2021 19:53:33 version 18.50.40.7.7.8.119 FWID 01-e6894c69
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: USB3.0 Hub
USB Device: AX88179
USB Device: USB2.0 Hub
USB Device: USB 2.0 BILLBOARD
USB Device: USB 3.1 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

```

mortbopet, 2021-07-19:
The crash on exit is most likely due to #103 - I'm still trying to figure that one out, it's one of those issues that are pretty difficult to debug without having the system at hand :thinking:.

Regarding compiler detection, have you then tried with a toolchain with multilib enabled? If you're using a prebuilt toolchain from https://github.com/sifive/freedom-tools note that there seems to be an issue with multilib support for recent toolchains (see https://github.com/sifive/freedom-tools/issues/81).

hossein1387, 2021-07-19:
I compiled with `multilib` flag and it works! thanks!

closed 2021-07-19 by hossein1387