mortbopet/Ripes #105: Using gcc directive in Ripes editor

hossein1387, 2021-07-19:
I am trying to simulate a simple code that I can compile with GCC but Ripes keeps giving me error:

```
__startup_code__:
    lui  a0, %hi(irq_handler)
    addi a0, a0, %lo(irq_handler )
    addi ra, sp, 0
    ret

```

it appears Ripes does not like `%hi(irq_handler)` syntax.  Outside of Ripes I can compile the code but since I am using the same compiler that Ripes is pointing to, why this piece of code is giving me error?

mortbopet, 2021-07-19:
Is this using the Ripes built-in assembler or through inline assembly, in C mode?

hossein1387, 2021-07-20:
I am using built-in assembler in Ripes

mortbopet, 2021-07-20:
The built-in assembler is separate from any compiler that is registered. The reason for this is portability, so as to not require everyone to download a toolchain to use the program.
Currently, the built-in assembler only implements a subset of the relocations available in GCC - namely `%pcrel_hi` and `%pcrel_lo` (see https://github.com/mortbopet/Ripes/blob/master/src/assembler/rvrelocations.h). However, `%hi` and `%lo` should be pretty straight forward to add, so I'll have a look later today.

mortbopet, 2021-07-21:
@hossein1387 support for these should have been added in 7b9c61f, and are available in the latest continuous release.

hossein1387, 2021-07-23:
@mortbopet that was fast! thanks a lot!

closed 2021-07-23 by hossein1387