mortbopet/Ripes #107: Build and publish for snapcraft

mortbopet, 2021-08-03:
Integrate building and publishing of snapcraft packages into CI when publishing new releases.
https://snapcraft.io/docs/qt5-kde-applications

tinywrkb, 2021-11-28:
[PoC Flatpak packaging](https://github.com/tinywrkb/flatpaks/blob/master/com.github.mortbopet.ripes/com.github.mortbopet.ripes.yaml), it's not ready for Flathub publishing, but it's a good start.  
It's missing AppStream appdata/metainfo, and also should be tested that everything works as expected.  
Also, it would be nice to have an install target, instead of manually copying the install assets.

mortbopet, 2021-11-28:
> PoC Flatpak packaging

Nice!

> it would be nice to have an install target, instead of manually copying the install assets.

Is the [current installation target](https://github.com/mortbopet/Ripes/blob/7c4296fd3d7b6cf2d32b4e29f18a9e88a8f118ad/CMakeLists.txt#L133-L135) insufficient for this? I'm wondering if whether setting `-DCMAKE_INSTALL_PREFIX=${FLATPAK_DEST}` in the `config-opts` would be sufficient?

tinywrkb, 2021-11-28:
> Is the [current installation target](https://github.com/mortbopet/Ripes/blob/7c4296fd3d7b6cf2d32b4e29f18a9e88a8f118ad/CMakeLists.txt#L133-L135) insufficient for this? 

Oh, I missed that, it's in master, but not in the latest tagged release. 

> I'm wondering if whether setting `-DCMAKE_INSTALL_PREFIX=${FLATPAK_DEST}` in the `config-opts` would be sufficient?

It's [being set automatically by flatpak-builder](https://github.com/flatpak/flatpak-builder/blob/e706b7df2834d1f7fe88ef801b9b578961492463/src/builder-module.c#L1775).

I updated the manifest to build from the latest commit, and the installation target works as expected.  
If there's nothing wrong with the app, then it just needs an AppStream appdata/metainfo, and a screenshot, and I think it's pretty much ready for Flathub.

mortbopet, 2021-11-28:
> Oh, I missed that, it's in master, but not in the latest tagged release.

I see, sorry about that. I foresee a new release coming within the next month or so.

> then it just needs an AppStream appdata/metainfo, and a screenshot, and I think it's pretty much ready for Flathub.

I've gone ahead and generated a file using https://www.freedesktop.org/software/appstream/metainfocreator/#/ available [here](https://github.com/mortbopet/Ripes/blob/master/appdir/usr/share/metainfo/io.github.mortbopet.ripes.metainfo.xml). Let me know if you need anything else from me :+1: .

tinywrkb, 2021-12-01:
I sent a #187 with changes to the metainfo file. Details in the commit message.

It's preferable that an app developer will maintain the Flathub packaging.  
I mean, I don't mind maintaining it, but it's much better for the users and the platform that app developers will be involved in the distribution of the app, and will build and test in Flatpak environment as part of the development. See more details [here](https://github.com/flathub/flathub/wiki/App-Submission#who-can-use-flathub).

Anyway, it's pretty much ready for submission, my branch for PR against the Flathub `new-pr` is [here](https://github.com/tinywrkb/flathub/tree/io.github.mortbopet.Ripes), and you're willing to submit the app, then you can pick it up from there.

It's obviously missing a Risc-V toolchain (riscv64 only?), I don't think it should block submission and publishing, but I'll leave the decision to you.  
I prefer not to add a toolchain directly to the app, as it seems like we're going to need to build from source, especially if we want to offer an `aarch64` host (by default, Flathub CI builds for both x86_64 and aarch64).  
I don't like the idea of building a toolchain in an app packaging, it's not the right place for it.

There are a few options here:  
* Build and package the toolchain as an app extension. This is pretty much wrong as it too wasteful building a whole toolchain for one app.
* Building and package the toolchain as Freedesktop SDK extension. This is the right way, but I would like for someone who knows more about the architecture to handle this. I've done something similar recently for AVR [here](https://github.com/tinywrkb/flatpaks/blob/master/org.freedesktop.Sdk.Extension.avr/org.freedesktop.Sdk.Extension.avr.yaml).
* Use a prebuilt toolchain and package it directly in the app. This might mean `x86_64` host only. In my very short test, I failed getting this to work on `x86_64`, I haven't investigated why.

tinywrkb, 2021-12-01:
I forgot to mention that I didn't add filesystem access permissions, as it seems like the app is using the File Portal, and doesn't need to open/save files from outside the sandbox without going through the file dialog.

It's not the best medium, but if anyone wants to test before a PR is sent to Flathub, then I'm attaching a Flatpak bundle.
[Ripes-999dc4e2e7c7bcb5bfec9504aac13fba11ebc59d.tar.gz](https://github.com/mortbopet/Ripes/files/7636603/Ripes-999dc4e2e7c7bcb5bfec9504aac13fba11ebc59d.tar.gz)

tinywrkb, 2021-12-03:
@mortbopet thanks for merging the metainfo changes.

I updated the [io.github.mortbopet.Ripes branch](https://github.com/tinywrkb/flathub/tree/io.github.mortbopet.Ripes) accordingly, and also dropped the unneeded cleanup, and added a `flathub.json` so PRs that are sent by flathubbot after it runs `flatpak-external-data-checker` (f-e-d-c) will not be automatically merged if the CI build run is successful.  
I squashed all my changes, but I kept the un-squashed commits in the [io.github.mortbopet.Ripes-old branch](https://github.com/tinywrkb/flathub/tree/io.github.mortbopet.Ripes-old).

Note that the way I set the `f-e-d-c`'s properties under `x-checker-data`, flathubbot will open a new PR automatically when a newer tag than `2.2.3` will be available, a tag property will be added, and the commit property will be updated.

tinywrkb, 2021-12-03:
We need to avoid using the File Portal when selecting the compiler with the file dialog.

```
diff --git a/src/settingsdialog.cpp b/src/settingsdialog.cpp
index ba8c263..3e6af80 100644
--- a/src/settingsdialog.cpp
+++ b/src/settingsdialog.cpp
@@ -177,6 +177,9 @@ QWidget* SettingsDialog::createCompilerPage() {
     auto* pathBrowseButton = new QPushButton("Browse");
     connect(pathBrowseButton, &QPushButton::clicked, [=, ccpath = ccpath] {
         QFileDialog dialog(this);
+#ifdef FLATPAK_TARGET
+        dialog.setOption(QFileDialog::DontUseNativeDialog);
+#endif
         dialog.setAcceptMode(QFileDialog::AcceptOpen);
         if (dialog.exec()) {
             ccpath->setText(dialog.selectedFiles().at(0));
```

mortbopet, 2021-12-03:
> flathubbot will open a new PR automatically when a newer tag than 2.2.3 will be available

Cool!

> We need to avoid using the File Portal when selecting the compiler with the file dialog.

Why is this?

tinywrkb, 2021-12-03:
The File Portal designed to give access to files outside the sandbox, the file dialog is run on the host, it has no access to files inside the sandbox (the mount namespace), only the file selected will be accessible, and its path will be a different one from where it's originated from (`$XDG_RUNTIME_DIR/doc/...`), see [here](https://flatpak.github.io/xdg-desktop-portal/portal-docs.html#gdbus-org.freedesktop.portal.FileChooser).

When we use the widget toolkit's file dialog (and not native), then we can select files from the sandbox environment/mount namespace.  

With the Flatpak `--filesystem` permission can bind mount folders (and files) from the host onto the same paths in the sandbox environment.

IIUC, in GTK3, the File Portal was not forced on apps, developers have to switch and use `FileChooserNative`.  
In Qt5, the File Portal is the default.

mortbopet, 2021-12-08:
@tinywrkb Set the flag in https://github.com/mortbopet/Ripes/commit/73d75685b45c73d4c1e3c5accbe891bc15a1157d. Would this not in theory be required for all file dialogs used in the program?

tinywrkb, 2021-12-08:
> Would this not in theory be required for all file dialogs used in the program?

It seems like it. Loading files (c source,...) through the File Portal works as expected, but not saving. I mean, it saves the file in the correct folder, but with the wrong filename, using a temporary filename instead.  
I'm guessing here (I didn't try to understand the code) that this is probably because the app uses the file dialog to read the file path, and then reuses this path, instead of opening the file directly for writing with the file dialog and using the same file descriptor.  

We should be able to avoid disabling the File Portal when saving and loading files, but this will likely require more changes to the app, so maybe take the easy route, and disable the File Portal for all the file dialogs.