mortbopet/Ripes #111: Issues with store operation?

hkarl, 2021-09-04:
Hi, 
it is probably a silly mistake, but I am struggling with the memory content of the memory after some store operation. E.g., ~sd~ should store 64 bits, right? So, test code: 

```
# lets put 0x0123456789ABCDEF into register x13
# (mabye there is a shorter way of doing that?)
lui t0, 0x01234
addi t0, t0, 0x567
slli t0, t0, 32
lui t1, 0x89ABD #-1 in next step
addi t1, t1, -529 #DEF
slli t1, t1, 32
srli t1, t1, 32
or x13, t0, t1
# x13 contains: 0x012345678ABCDEF ; all good 

# write that to memory ; let's be double word aligned for simplifity: 

sd x13, 0x40(zero)  
# Excpeted:          40:EF, 41:CD, 42:AB, 43:89, 44:67, 45:45, 46:23, 47:01
# Result on Mac:     40:EF, 41:CD, 42:AB, 43:89, 44:00, 45:00, 46:00, 47:00
# Result on Windows: 40:EF, 41:CD, 42:AB, 43:89, 44:00, 45:00, 46:23, 47:01
```

In particular, memory content differs when running on Mac and windows ? 

Any pointers much appreciated! 
best, HK

mortbopet, 2021-09-04:
Thank you for submitting the issue; there should obviously never be cross platform discrepancies between program executions so if that is the case, something is definitely wrong - i'll have a look into it.

mortbopet, 2021-09-04:
As for generating instruction sequences for loading large constants, this is what the `li` pseudo-instruction is for. I.e.: `li x13 0x0123456789ABCDEF` should do the trick. 

This also seems to be an issue on Linux. Upon a b it of digging, this seems to be an issue with the memory viewer itself, and not whats going on in the simulator - executing the following prints the expected value:
```asm
li x13 0x0123456789ABCDEF
sd x13, 0x40(zero)  
ld x14 0x40(zero)
li a7 34
mv a0 x14
ecall
```

mortbopet, 2021-09-04:
Once available, please have a look at the [continuous release](https://github.com/mortbopet/Ripes/releases/tag/continuous) to see whether the issue has been fixed. I'll some point (soon) make a new minor release containing bug-fixes related to the 64-bit processors, so if you do happen to stumble across anything else, please do not hesitate to let me know!

nkluge, 2021-09-06:
Hi,
I'm working with @hkarl. I have supplied the Windows output. To be honest, I cannot reproduce it. So either is was a glitch on my machine or I just did a copy-paste mistake (more likely). I really don't know. So for now we can assume that the Windows and Mac output looks the same. Sorry for the inconvenience.

I've tested your [code](https://github.com/mortbopet/Ripes/issues/111#issuecomment-912957145) and in fact it was just a bug in the memory viewer. The result of your code looks fine (Windows). The continuous release fixes this visual bug for me (Windows). The content is displayed correctly.

Thanks a lot for the fast reply and fix!
Best, Norman

closed 2021-09-06 by mortbopet

hkarl, 2021-09-07:
Thanks a lot; continuous also works as expected on macOS. Wonderful support!