mortbopet/Ripes #123: Add example program for file I/O

mortbopet, 2021-09-29:
There should be an example program to demonstrate the use of the various file-related system calls.
The example program could:
- open a new file for writing
- write something to the file
- seeking in the file
- closing the file
- reopening the file and printing the contents of the file
- ....

jakariaaa27, 2022-01-05:
Hi @mortbopet , let me introduce myself, my name is jakaria. I have a final project at my university. one of the topics in this final project is to contribute to solving the issues that exist in your project. I want to try to help you in this case. what I understand in this case you want to add a new various to the system call so that it can help the user if needed. here you want to add
* open a new file for writing
* write something to the file
* seeking in the file
* closing the file
* reopening the file and printing the contents of the file
can you explain more detail about it?
what I know in the system call in this latest version only briefly describes a function. examples like PrintInt, PrintFloat, Exit... etc.

jakariaaa27, 2022-01-06:
@mortbopet do you mean if we click "open" on the system call a dial dialog will appear showing, flag argument? then inside the dial dialog, we can write something inside the file and we can edit the dial dialog for everything?

mortbopet, 2022-01-06:
Hey @jakariaaa27 - What is needed for this issue is an example assembly language program (like the others in `file->load example->assembly->...` which demonstrates the use of the file IO system calls. As i think you hinted to (and as hinted to by the original issue #121), the system call help dialog at `help->system calls` provides info of the available system calls ---including the file IO related ones--- but this might be a bit ambiguous for those who have not worked with low-level file APIs before.

In writing such an example assembly program, i would suggest you to first sketch out something in C where you also use the linux file API function like [fstat](https://linux.die.net/man/2/fstat), [lseek](https://linux.die.net/man/2/lseek), [close](https://linux.die.net/man/2/close) and so forth, covering all of the file-related system. Once this C program works, you can then consider how to rewrite it in assembly using the system calls available in Ripes.

jakariaaa27, 2022-01-06:
okay, I got it. so is mean if we open file->load example->assembly->led.s the file is will open inside the editor and we can modify and run the program. is it like you mean?

jakariaaa27, 2022-01-06:
okay, I got it. so is mean if we open file->load example->assembly->led.s the file is will open inside the editor and we can modify and run the program. is it like you mean? because now if we try to open file->load example->assembly->led.s, we cant open the file in the editor.

jakariaaa27, 2022-01-06:
are you expecting the --help to show the information of all system call and implement how these systems can be used and show the example?

Sreeram-Ramesh, 2025-04-21:
Hi @mortbopet - I am Sreeram, and I am currently in my junior year. I use Ripes for a course of mine and would love to contribute! 

In my course, I have had assignments on programming using RISC-V assembly, and I have dealt with file handling in C. As you said I can try with fstat, lseek and close. I am new to contributing, and have some doubts:

> 1. So I should write a single program that does all 5 listed right?

> 2. Once I get the C code and the assembly code where should I add those in this repo?

mortbopet, 2025-04-22:
@Sreeram-Ramesh 

> 1. So I should write a single program that does all 5 listed right?

I think a single would be fine.

> 2. Once I get the C code and the assembly code where should I add those in this repo?

If you search for one of the existing example files (as they are named in Ripes), that should point you directly to the location :). https://github.com/mortbopet/Ripes/tree/master/examples

Sreeram-Ramesh, 2025-04-22:
@mortbopet 

So this is the C code I wrote:

```C
#include<stdio.h>
#include<unistd.h>
#include<sys/stat.h>
#include<sys/types.h>
#include<fcntl.h>

int main() {

    // File created in write mode.
    int fptr = open("sample.txt", O_WRONLY | O_CREAT, 0644);
    char buff[33] = "Hello Ripes! This is a test file.";
    if (fptr < 0) {
        printf("Error in writing!\n");
    }
    write(fptr, buff, sizeof(buff));
    printf("Written successfully!\n");

    // File stats diplayed here using fstat.
    struct stat file_info;
    int status = fstat(fptr, &file_info);
    if (status < 0) {
        printf("Error in accessing the file info!\n");
    }

    off_t file_size = file_info.st_size;
    uid_t user = file_info.st_uid;
    dev_t device = file_info.st_dev;
    mode_t file_mode = file_info.st_mode;

    printf("File size: %ld bytes\n", file_size);
    printf("User Id: %d\n", user);
    printf("Device: %ld\n", device);
    printf("File mode: %d\n", file_mode);

    close(fptr);

    // File opened in read.
    fptr = open("sample.txt", O_RDONLY);
    int bytes = 30;
    char read_buff[bytes];
    if (fptr < 0) {
        printf("Error in reading!\n");
    }
    read(fptr, &read_buff, bytes);
    printf("Read %i bytes: %s\n", bytes, read_buff);

    // Using lseek to change the cursor position.
    lseek(fptr, -20, SEEK_CUR);
    read(fptr, &read_buff, bytes);
    printf("Read after lseek: %s", read_buff);

    close(fptr);

    return 0;
}
```

Any suggestions? Would like your feedback before l move on with this!

Sreeram-Ramesh, 2025-04-29:
Hey, so I initially tried to build the toolchain for compiling the C example files but then met with submodule error,  and I couldn't resolve it so I then moved on to pre-built toolchains from SiFive, the one exactly mentioned in ```c_programming.md```, added ```-nostdlib``` to linker arguments and then tried executing examples that's when only ```matrixmul.c``` file worked as others had stdlib functions.

Now I have few doubts:

> 1. Should I remove stdlib functions from my C file and replace them with ecall?

> 2. If I am able to build assembly files using the compiler should I still write the assembly code example file?

Just thought will let you know that location of RISCV C/C++ compiler setting needs change in documentation and dialog boxes to ``` Edit -> Settings -> Compiler ``` from ``` Edit -> Settings -> Editor ```.

Have a great day! :)

Sreeram-Ramesh, 2025-05-05:
Hey, so everything seems to work fine except getting the FStat right, it's always zero at the memory, here's the code in assembly:

```asm
# This example shows how a file can be opened in write
# mode, content written into the file, seek operation
# performed in file, close operation performed on file,
# and finally opening the file in read mode and reading
# from the file.

.data
filename:       .string       "sample.txt"
newline:        .string       "\n"
content:        .string       "This is a sample text file. Hello ripes!\0"
fstat_struct:   .zero         144

.text
# Creating a file
la a0, filename
li a1, 1
li a7, 1024
ecall

mv t0, a0 # file descriptor - t0

# Writing to the file
la a1, content
li a2, 41
li a7, 64
ecall

# Fstat to get file information
mv a0, t0
la a1, fstat_struct
li a7, 80
ecall

# Closing the file
li a7, 57
ecall

# Standard exit from file
li a7, 10
ecall

```

Am I missing something?
Thanks!

Sreeram-Ramesh, 2025-05-27:
Hey @mortbopet , hope you are having a good day.

This is what I found regarding fstat in ripes, I was scratching my head why it always returned null values even when no errors to find this:

[image]

So fstat seems still not complete, and null values show that the syscall seems to be working fine but under current development not to the fullest.

So with that, here's the code incorporating all the system calls you asked for. If you are satisfied with this I can proceed with PR.

```asm
###########################################################
# This example shows how a file can be opened in write
# mode, content written into the file, seek operation
# performed in file, close operation performed on file,
# and finally opening the file in read mode and reading
# from the file.
#
# Note: Check the current directory as ripes for the
# sample.txt file, also delete the file for the program
# to execute properly again.
###########################################################

.data
filename:       .string       "sample.txt"
content:        .string       "This is a sample text file. Hello ripes!"
newline:        .string       "\n"
display:        .string       "File size: "
fstat_struct:   .zero         128
readbuf:        .zero         64

.text
# Creating a file
la a0, filename
addi a1, zero, 0x242 # O_RDWR | O_CREAT | O_TRUNC
addi a7, zero, 1024
ecall

add t0, zero, a0 # file descriptor - t0

# Writing to the file
add a0, zero, t0
la a1, content
addi a2, zero, 41
addi a7, zero, 64
ecall

# LSeek to change the pointer to beginning
add a0, zero, t0
add a1, zero, zero
add a2, zero, zero
addi a7, zero, 62
ecall

# Reading from the file
add a0, zero, t0
la a1, readbuf
addi a2, zero, 41
addi a7, zero, 63
ecall

# Print the read content
la a0, readbuf
addi a7, zero, 4
ecall

# Closing the file
add a0, zero, t0
addi a7, zero, 57
ecall

# Fstat to get file information
add a0, zero, t0
la a1, fstat_struct
addi a7, zero, 80
ecall

# Printing the file size
la a0, newline
addi a7, zero, 4
ecall

la a0, display
addi a7, zero, 4
ecall

la a5, fstat_struct
lw a0, 48(a5) # As per Linux ABI offset 48 is file size
addi a7, zero, 1
ecall

# Note: As of 2025, the fstat syscall populates the 
# structure with just zeros, hence the file size 0.

# Standard exit from file
addi a7, zero, 10
ecall
```

These are the outputs I get:

Ripes Console:
[image]

The text file created:
[image]

Thanks ;)

Sreeram-Ramesh, 2025-07-25:
@mortbopet No problem, I have made a PR a while ago, #409 , I had some formatting issues which is resolved now. Have a great day!