mortbopet/Ripes #125: [RVC][Tracker]

mortbopet, 2021-10-02:
This is a tracking issue for tracking the implementation of the RISC-V compressed ISA extension in Ripes.

- [ ] https://github.com/mortbopet/Ripes/issues/124

Discussion on support for the C extension: https://github.com/mortbopet/Ripes/discussions/119

closed 2021-11-16 by mortbopet

Zikon328, 2025-02-03:
Disassembler does not correct dissasemble register, when reduce range register x8-x15
and not error for  reduce reg in asm source.
( Asm and C  mode )
[image]

Command  c.beqz  and  c.bnez   not  valid  compute  offset.
( Asm mode )

format command   c.lw and  c.sw
c.lw  rd,  rs,  offset

not really   
c.lw   rd,  offset(rs)

Xiamatsu, 2025-06-20:
Branch and jump commands "C" instructions is not valid offset !
c.beqz c.bnez
c.j c.jal
The absolute address is used, not relative.

Commands c.lw and c.sw - used only format
c.lw rd, rs, offset

can do [ c.lw rd, (offset)rs ] ?