mortbopet/Ripes #128: [RVC] Add processor support for the C extension

mortbopet, 2021-10-02:
In enabling processor support for the C extension, we need to:

- Modify the instruction decode unit to expand an instruction from its compressed to its uncompressed counterpart.
- This will also have an influence on the path which we forward the value of an instruction to i.e. the `ImmGen` component. Currently, these are hooked **directly** to the value of the read instruction coming out of the `IFID` stage. Instead, they need to be hooked up to the expanded instruction, presumably as an output of the modified decode unit.

mortbopet, 2021-10-02:
Discussion on this at https://github.com/mortbopet/Ripes/discussions/119

lcgamboa, 2021-10-03:
I have added a PC increment output (2 or 4) and an expanded instruction output to link directly to Imm.Gen. It worked, but the diagram was left with messy connections. 

[image]

I would like to know if it is possible to use any method of the Assembler class to assemble one single instruction. This is to be used in the expansion of the instructions. 
For example the instruction `C.LI rd, imm[5:0]` has to be expanded to `addi rd,x0, imm[5:0]`. 
I'm doing the expansion with the code:
`new_instr = (imm << 20)|(rd<<7) |RVInstr::ADDI;`
but I believe I have a better way of doing it using existing resources.

mortbopet, 2021-10-04:
> It worked, but the diagram was left with messy connections.

The work on improving placement and routing algorithms in VSRTL has been inactive for a while, and as such, to get "pretty" drawings we essentially have to manually adjust the wires and sizes of the components in the drawing. I can recommend to postponing this until you know that everything is working as expected, and all tests pass - it can be a bit time consuming to get everything to line up!

> I would like to know if it is possible to use any method of the Assembler class to assemble one single instruction. This is to be used in the expansion of the instructions.

What you're looking for here is the support for pseudo-instructions. Pseudo instructions are shorthand instructions which map to one or multiple other instructions. Support for this can be implemented like:
https://github.com/mortbopet/Ripes/blob/651bf82b5475c99131c8a32681e36598b7a1437f/src/assembler/rv_i_ext.h#L90-L93
where we define a replacement set of tokens to be emitted instead of the matched instruction.

If `c.li` has [the same complexity as the `li` pseudoinstruction](https://github.com/mortbopet/Ripes/blob/master/src/assembler/rv_i_ext.h#L171), then i can recommend you to look up the corresponding implementation in i.e. GCC or LLVM, things they've worked out all of the kinks and edge-cases of the instructions, and the API for expanding pseudo instructions in LLVM translated fairly smoothly to what is done in Ripes.

 I was wondering whether, for purely didactic reasons, if it would be better to have the PC increment output of the decode unit as a multiplexer select signal, instead of the increment value itself and have PC incrementation be organized like this:
[image]
This would give a nice visual hint that we're executing compressed instructions... just a thought!

lcgamboa, 2021-10-05:
I added the multiplexer as you suggested. The problem with messy connections in the diagram is because I haven't modified the layout.json files (is it possible to edit json with VSRTL?), but this problem is just visual and can be fixed later. 

[image]

My initial idea was to use edittab's assembler and disassembler to do the decoding (disassembler (RVC16)->tokens->16 to 32 conversion -> tokens-> assembler(RVI)). But apparently it's only prepared to work with files. As the processor decoding is done separately in the [riscv.h](https://github.com/mortbopet/Ripes/blob/1fffcb43cf0ad38e2edffd958e79727e4e24ec47/src/processors/RISC-V/riscv.h#L57) file, I will follow the same path.

mortbopet, 2021-10-05:
> is it possible to edit json with VSRTL?

If you right click outside of the processor you can select "unlock". This will allow you to move and adjust the components themselves, add new wire points (right-click on wires), collapse wire points onto each other (drag adjacent points from the same wire on top of each other), hide and show labels and ports, and more.

> My initial idea was to use edittab's assembler and disassembler to do the decoding

Correct, the disassembler/assembler is right now separate from the processor implementations, and the processor decode unit uses the method that you found, i.e. the `generateInstrParser` to split an input instruction into its various fields.

mortbopet, 2021-10-30:
5 and 6 stage support is added in 22603b3. I don't see a compelling argument for also adding compressed support to the "intermediate" 5-stage models, so i think we can consider this task finished.

closed 2021-10-30 by mortbopet

lcgamboa, 2021-10-30:
> 5 and 6 stage support is added in [22603b3](https://github.com/mortbopet/Ripes/commit/22603b3c2b87ff0ec933a8edab17d7dac122113d). I don't see a compelling argument for also adding compressed support to the "intermediate" 5-stage models, so i think we can consider this task finished.

I've already added support for all processors, including intermediate ones.