mortbopet/Ripes #130: [RVC] Implement all "C" instructions

mortbopet, 2021-10-02:
Currently, only a subset of the compressed instructions are implemented:
https://github.com/mortbopet/Ripes/blob/957eb95925397dc6a99100d05a7d0ecc270a329d/src/assembler/rv_c_ext.h#L30-L35

The remainder of the ISA extension should also be implemented. Refer to page 82 of https://riscv.org/wp-content/uploads/2017/05/riscv-spec-v2.2.pdf for the full list.

lcgamboa, 2021-10-03:
I'm adding CI type instructions with the macro below: 
```
#define CIType(opcode, name, funct3)                                                              \
    std::shared_ptr<_Instruction>(new _Instruction(                                               \
        Opcode<Reg__T>(name, {OpPart(opcode , 0, 1), OpPart(funct3, 13, 15)}), \
        {std::make_shared<_Reg>(isa, 1, 7, 11, "rd/rs1"), \
         std::make_shared<_Imm>(                                                                         \
             2, 6, _Imm::Repr::Signed,                                                                  \
             std::vector{ImmPart(5, 12, 12), ImmPart(0, 2, 6)})}))    
```
But there are some instructions with the same opcode that only use rd ==0 or rd != 0 to do the mapping.  As c.nop and c.addi: 

[image]
Apparently the statements below would have the opcode duplicated.
```
       instructions.push_back(CIType(0b01, Token("c.addi"), 0b000));
       instructions.push_back(CIType(0b01, Token("c.nop"), 0b000));
```
Which I think would bring problems. But I still have no idea how to separate the instructions in this case using this kind of macro. 
Any idea how to do this?

mortbopet, 2021-10-04:
In this case, i would probably define the 7-11 bits of the `c.nop` instruction to be part of the `Opcode` field of the instruction. The intent of the `Opcode` field is essentially to match on all constant parts of an instruction - which would be the case for the 7-11 bits of `c.nop`.

If doing the above, then you might run into an issue in the [instruction matcher](https://github.com/mortbopet/Ripes/blob/651bf82b5475c99131c8a32681e36598b7a1437f/src/assembler/matcher.h#L82) used for disassembly, complaining about insufficient discernible parts of the instruction, i.e., that `c.addi` and `c.nop` might alias since `c.addi`s opcode is identical to a subset of `c.nop`. If this is the case, then we should modify the matcher to handle support for the fact that `c.nop` and `c.addi` in fact _are_ discernable since it is only a _subset_ of `c.nop` which aliases with `c.addi`.

lcgamboa, 2021-10-05:
Your proposal solved the problem, I added a specific definition for the NOP. Thanks for the tip. 

```
#define CIType(opcode, name, funct3)                                                              \
    std::shared_ptr<_Instruction>(new _Instruction(                                               \
        Opcode<Reg__T>(name, {OpPart(opcode , 0, 1), OpPart(funct3, 13, 15)}), \
        {std::make_shared<_Reg>(isa, 1, 7, 11, "rd/rs1"), \
         std::make_shared<_Imm>(                                                                         \
             2, 6, _Imm::Repr::Signed,                                                                  \
             std::vector{ImmPart(5, 12, 12), ImmPart(0, 2, 6)})}))    
    
#define CINOPType(opcode, name)                                                              \
    std::shared_ptr<_Instruction>(new _Instruction(                                               \
        Opcode<Reg__T>(name, {OpPart(opcode , 0, 1), OpPart( 0 , 2, 15)}),{}))  
```

mortbopet, 2021-10-05:
4330e57 adds support for the compressed register notation.

commit c8ad2247 (2021-10-12): Implement all C instructions (#133)

* implement all C instructions

closes #130 .