mortbopet/Ripes #131: Console view in editor mode

schoeberl, 2021-10-04:
Single stepping a program in editor mode is a nice view for using Ripes as an ISA simulator (without looking into pipeline details). It would be nice if this view can also show the console output.

mortbopet, 2021-10-04:
Added a setting for this in #1fffcb4 - you can test it out in the [latest continuous build](https://github.com/mortbopet/Ripes/releases/tag/continuous). By now, it's off by default - let me know if this is something which you'd expect to be enabled by default.

schoeberl, 2021-10-04:
I would like to see Ripes also as a plain ISA simulator (to drop Venus), which it is when selecting the single-cycle processor and stay in the editor view. Maybe this should then be on by default.

mortbopet, 2021-10-05:
Agreed. 3c98bb0 enables the console by default.

closed 2021-10-05 by mortbopet