mortbopet/Ripes #166: Disassembler does not correctly disassemble some RVC instructions

lcgamboa, 2021-10-30:
Some RVC instructions do not disassemble correctly: c.addi, c.lui, c.li, c.mv, c.srli and c.jal.
An example program below. The program works perfectly, the problem is only in the disassembly. 
```
c.addi a0, 1
c.lui a1, 0x2000
c.li a2,3
c.mv a3,a1
c.srli a3,3
c.jal -10
```
[image]
`

mortbopet, 2021-10-30:
Working on it at the moment; there are indeed some issues in both how bitfields are compared together as well as how the instruction matcher generates its match tree. Interesting that this has been highlighted through the C extension but wasn't an issue in the other parts of the ISA!

lcgamboa, 2021-10-30:
I believe it is because for normal ISA the opcode is 7 bits and it is always set from bit 6 to 0, which is not the case for RVC.

mortbopet, 2021-10-30:
Should hopefully be fixed in the above set of commits...

lcgamboa, 2021-10-30:
With this example it worked, but loading the `test/riscv-tests-c/rvc.s` file Ripes freeze. 

Starting Ripes with the program `test/riscv-tests-c/rvc.s`  loaded, the following error is displayed: 
```
Ripes: ../Ripes/src/assembler/program.cpp:44: void Ripes::DisassembledProgram::set(unsigned int, Ripes::VInt, const QString&): Assertiva “addressToDisresMap.count(address) == 0” falhou.
Abortado
```

mortbopet, 2021-10-30:
yep, found that as well, and trying to fix it know... It's like playing whack-a-mole!

lcgamboa, 2021-10-30:
Everything working, problem solved.

closed 2021-10-31 by mortbopet