mortbopet/Ripes #188: Ripes - web edition (Qt for WebAssembly)

mortbopet, 2021-12-03:
This issue is intended to track progress on getting a web-based version of Ripes up and running. This would be based on [Qt for WebAssembly](https://doc-snapshots.qt.io/qt6-dev/wasm.html). https://mortbopet.github.io/ already shows a very old, very broken version of Ripes running on a very old version of Qt for WebAssembly.

Below are the main issues with getting things up and running for the current version of Ripes:
- Proper support for threads/futures in WebAssembly. This would include support for the Qt thread/future modules, and not just webassembly support.
- Support for CMake-based Qt for WebAssembly builds
- Support for all of the Qt libraries that we rely on:
  - [x] QtSvg
  - [ ] QtCharts

A good way to progress on this would be to set up a CI job on a separate branch which
- fetches prebuilt versions of the current state-of-the-art Qt+WebAssembly compilers
- configures Ripes CMake for webassembly
- Builds, and lets us know where things failed/improved

erickuo5124, 2022-01-19:
Hi @mortbopet !
I'm currently working on this issue, and hope to see that Ripes can be easily access by browser. I can successfully build the Desktop version of Ripes, but failed on WebAssembly version. 

Below are the error messages:

```
[  8%] Building CXX object external/VSRTL/graphics/CMakeFiles/vsrtl_gfx_lib.dir/vsrtl_gfx_lib_autogen/mocs_compilation.cpp.o
In file included from /home/kuouu/Ripes/external/VSRTL/graphics/vsrtl_gfx_lib_autogen/EWIEGA46WW/moc_vsrtl_widget.cpp:10,
                 from /home/kuouu/Ripes/external/VSRTL/graphics/vsrtl_gfx_lib_autogen/mocs_compilation.cpp:15:
/home/kuouu/Ripes/external/VSRTL/graphics/vsrtl_widget.h:8:10: fatal error: QtConcurrent/QtConcurrent: No such file or directory
    8 | #include <QtConcurrent/QtConcurrent>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [external/VSRTL/graphics/CMakeFiles/vsrtl_gfx_lib.dir/build.make:103: external/VSRTL/graphics/CMakeFiles/vsrtl_gfx_lib.dir/vsrtl_gfx_lib_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:864: external/VSRTL/graphics/CMakeFiles/vsrtl_gfx_lib.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

I use the latest version of Ripes with Qt5.15 and emscripten 1.39.8, and it seems that QtConcurrent is not support in this version of Qt for WebAssembly. Besides, I afraid of the error occur because I'm not familiar with cmake, so I try [this docker image](https://hub.docker.com/r/reivilo1234/qt-webassembly-cmake) (for CMake based C++/Qt5 application) and get the same error. I also download Qt6.12 and can not find QtConcurrent header file either.

In addition, current version of Ripes failed to build for Qt5.14, but readme say that 5.14+ version of Qt. It took me a long time to figure out the problem😂😂

mortbopet, 2022-01-19:
Hi @erickuo5124, thank you for taking a look at this.

Yea, the main blocker for getting Ripes to build in a webassembly version as always been the support for Qt's multithreading features. This is something that is actively being worked on by Qt, but it seems like there might soon be enough support for things to work. Qt sent out an update wrt. Qt6.2+WebAssembly support. Seems like they have gotten QCharts supported now! https://www.qt.io/blog/qt-6.2-web-assembly-whats-new?utm_campaign=Qt%206&utm_content=194270879&utm_medium=social&utm_source=linkedin&hss_channel=lcp-4788303. At the bottom of that page, they mention some new features for Qt6.3 + multithreading.

> In addition, current version of Ripes failed to build for Qt5.14, but readme say that 5.14+ version of Qt. It took me a long time to figure out the problemjoyjoy

Should probably bump this up to 5.15 then, thank you for catching it.

magiwanders, 2023-01-18:
I have returned to this issue on my local version of the repository with the following setup:

- QtCreator 9.0.1-1
- WebAssembly Qt 6.5.0 beta-1 (Multi-Threaded) AND WebAssembly Qt 6.6.0 dev snapshot (Multi-Threaded) 
- Emscripten 3.1.13
- Ripes updated to the latest commit
- VSRTL updated to the state described in mortbopet/VSRTL#55 to rule out errors coming from its module

I have found that all necessary support for cmake, QtCharts, QtConcurrent (multithreaded) is there, but some compile time errors still occur. What follows is a roadmap for tracking progress:

- [x] 1. Migrating VSRTL to Qt6.5 (see mortbopet/VSRTL#55) - This seems to be mostly done.
- [ ] 2. Migrating Ripes to Qt6.5 (see mortbopet/Ripes#262) - This compiles but has substantial and breaking runtime errors.
- [ ] 3. Compiling VSRTL to WebAssembly (see ISSUE_TODO) - This still spits out a bunch of errors at compile time of the following kinds:
>wasm-ld: warning: function signature mismatch: time

This laments that something is defined as i32 in Qt and i64 in emscripten. This is automatically resolved using Qt 6.6.0 dev snapshot. Other errors are of the type:

>wasm-ld: error: /home/magiwanders/Qt/6.5.0/wasm_multithread/lib/libQt6Core.a(qtenvironmentvariables.cpp.o): undefined symbol: std::__2::__basic_string_common<true>::__throw_length_error() const
>wasm-ld: error: /home/magiwanders/Qt/6.5.0/wasm_multithread/lib/libQt6Core.a(qurlidna.cpp.o): undefined symbol: std::__2::__basic_string_common<true>::__throw_out_of_range() const

The errors above persist in Qt 6.6.0

- [ ] 4. Compiling Ripes to WebAssembly (see ISSUE_TODO) - This spits out virtually identical errors as the previous point.

mortbopet, 2023-01-18:
@magiwanders This sounds like great progress! I'm super excited about this, and it sounds like we aren't too far off from being able to build a wasm version of Ripes! Let me know if there is anything i can help you with apart from the issues that you've raised in VSRTL/Ripes.

Regarding Qt+wasm compilation issues, if it doesn't "just work", i'd suggest raising them on the Qt boards since those errors are most likely not be Ripes-specific.

mortbopet, 2025-01-05:
Closing since https://ripes.me/ went live some time ago.

closed 2025-01-05 by mortbopet