mortbopet/Ripes #191: Can't v2.2.3 release on Arch from git through AUR or manually

ekliptik, 2022-01-03:
Manually:
```
CMake Error at external/CMakeLists.txt:11 (add_subdirectory):
  The source directory

    /home/emil/pulls/Ripes-2.2.3/external/VSRTL

  does not contain a CMakeLists.txt file.
```
Through the AUR package [ripes-git](https://aur.archlinux.org/packages/ripes-git/):
```
/home/emil/.cache/yay/ripes-git/src/ripes-git-2.2.3/external/libelfin/elf/mmap_loader.cc:15:10: fatal error: external/cpp-mmaplib/mmaplib.h: No such file or directory
   15 | #include "external/cpp-mmaplib/mmaplib.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
```
cmake 3.22.1
g++ (GCC) 10.2.0
Qt 5.15.2 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 11.1.0) on "xcb"
OS: Arch Linux [linux version 5.15.11-arch2-1]
```

mortbopet, 2022-01-04:
For the first one, are you sure that you recursively pulled submodules when you checked out Ripes?
The second one also seems a bit strange; v2.2.3 doesn't use `libelfin` so unsure why that is even part of the tree that has been checked out, hmm...

ekliptik, 2022-01-04:
My bad, should have read the readme more diligently, recursive clone fixed it.
A couple days ago I have seen a repository that used submodules but pulled them in automatically, but I can't figure out how to do that. I'll take a look at the AUR pkgbuild and see if I can fix that next

ekliptik, 2022-01-04:
I tried to fix the PKGBUILD on my own but couldn't figure it out so I left a comment for the repo maintainer [here](https://aur.archlinux.org/packages/ripes-git/#comment-844843)

ltoenning, 2022-01-04:
Thanks for reporting. This should be fixed with the newest AUR package update.

@mortbopet: It is pulling the ``libelfin`` submodule because it is a ``-git`` AUR package which always builds from upstream/master. Unfortunately AFAIK these git-packages still need a version number for AUR... That's why it is labeled as v2.2.3...

ekliptik, 2022-01-05:
Thank you, works on my machine, closing. AUR really could do with a rethinking of versioning for git packages, I agree. By the way, Ripes made it really fun and easy to debug my single cycle core, and I ran my zig emitted ELF binary on Ripes with no problems, good times.

closed 2022-01-05 by ekliptik