mortbopet/Ripes #199: local labels not working ?

hl037, 2022-03-15:
Hi,
From here : https://github.com/riscv-non-isa/riscv-asm-manual/blob/master/riscv-asm.md#labels I is said we can use numeric local labels...

...But I get an error when trying to do so... Is it expected ?

[image]

mortbopet, 2022-03-15:
Hi @hl037 

That part of the assembly spec has not been implemented in the Ripes assembler. It shouldn't be too hard to implement though; I'll take a look at it. Thank you for raising the issue!

mortbopet, 2022-03-17:
@hl037 took a stab at writing an implementation for this - the feature should be available in the [continuous release](https://github.com/mortbopet/Ripes/releases/tag/continuous). Please let me know if it works for you.

closed 2022-03-22 by mortbopet