mortbopet/Ripes #207: Ripes won't open up

joaomiguelaj, 2022-05-15:
Hello, I installed ripes because I need it in college. For context, my OS is Windows11.
I installed Visual C++ and I installed Ripes, but when I unzip it and try to open the application, I click it twice, but nothing happens, it shows on the mouse pointer that is loading but nothing opens.

I need your help, please!
Thank you.

mortbopet, 2022-05-16:
Hi @joaomiguelaj 

Which version are you running? If you open Ripes through a shell (i.e. open Ripes through PowerShell) is there any output printed in the terminal?
I've yet to do any testing on Windows 11, so that might also be a reason for the issues...

joaomiguelaj, 2022-05-16:
> Hi @joaomiguelaj
> 
> Which version are you running? If you open Ripes through a shell (i.e. open Ripes through PowerShell) is there any output printed in the terminal? I've yet to do any testing on Windows 11, so that might also be a reason for the issues...

Kecxelic, 2022-06-08:
I have the same problem,I can't open it,too.And my OS is also Windows11.

Kecxelic, 2022-06-08:
> Locally I'm able to open v2.2.4 it on Windows 11, so i cannot reproduce the issue.
>  To get to the core of this, it might be good if you could try building Ripes on your own machine(s) and then running it in Debug mode... Hopefully that should give us more clues as to what is going on.

Thank you for your letter. I just became a student of Computer Science ​for one year. I'm afraid I don't have enough ability to compile and run by myself. Now I'm trying to use the rsic-v support plug-in in vscode, I'm sorry that I can't find out the problem.
In addition, I am really happy to be answered by an excellent  programmer.

mortbopet, 2025-01-05:
Closing due to inactivity.

closed 2025-01-05 by mortbopet

MiguelRuiz683, 2025-10-27:
I have the same problem. Sometimes, it spawns a window saying that colud be dangerous, but I click run anyway and it doesn't open.  Other times, directly nothing happens