mortbopet/Ripes #210: out of order riscv

roua987, 2022-06-15:
how i could  add processor architecture that i want ?

mortbopet, 2022-06-15:
Hi @roua987

Out of order processors are fairly complex. While i've thought about adding this before, i've yet to allocate the time and resources to do it.

One issue with this is that, while VSRTL (the language which Ripes' processors are described in) is fairly capable, I fear that implementing an OoO processor would be _very_ cumbersome if the language is not extended in some shape or fashion.

I think a better approach would be to write an implementation of your processor in i.e. SystemVerilog, and then attach it to Ripes through verilator - this is experimental, but i see it as a necessary approach for more complex processors, see https://github.com/mortbopet/Ripes/blob/master/docs/new_processor_models.md#experimental-verilator-processor-models-in-ripes.

mortbopet, 2025-01-05:
Closing due to inactivity.

closed 2025-01-05 by mortbopet