mortbopet/Ripes #212: Non-stop opening

chuanEgg, 2022-07-05:
My OS is ubuntu 20.04.
I installed the AppImage file and used `chmod a+x Ripes-v2.2.4-linux-x86_64.AppImage` and `./Ripes-v2.2.4-linux-x86_64.AppImage` to run the app. However, it lagged for a while, and suddenly multiple Ripes opened up and didn't stop until a reboot my system. In addition, many `AppRun` can be sighted in the System Monitor. I think I need help, and I apologize for my poor English.
[image]
here's a screenshot of the chaos.

mortbopet, 2022-07-05:
Hmm, very strange! i run ubunt 20.04 myself and have never had this happen. What happens if you try to unpack the appimage and run the binary directly?
```sh
./Ripes-v2.2.4-linux-x86_64.AppImage --appimage-extract
mv squashfs-root/usr/bin/Ripes squashfs-root/usr/lib/
cd squashfs-root/usr/lib/
./Ripes
```

chuanEgg, 2022-07-06:
That worked! Thank you very much. However, I'm still very curious on the subject, perhaps I will spend some time figuring it out.

closed 2022-07-06 by chuanEgg