mortbopet/Ripes #223: Search in the code and debug point

xdelu02, 2022-09-01:
It will be nice if you could search words in the code with ctrl + f like in all text editors. 

Also why i can put the debug point only on the disassembled code, makes no sense for me

mortbopet, 2022-09-02:
> It will be nice if you could search words in the code with ctrl + f like in all text editors.

That would be a nice feature, i agree!

> Also why i can put the debug point only on the disassembled code, makes no sense for me

When breakpoint support was added in Ripes, there was no notion of source mappings (i.e. which input assembly/C line corresponds to a given line in the assembled program). These source mappings are what IDEs/debuggers use to know when to pause execution if a given source line was hit (=> a specific line in the binary).

However, now, we in fact do have these source mappings (used for the stage highlighting), so there are in theory no major technical barriers to implementing what you suggest.

xdelu02, 2022-09-02:
Very nice, thank you!

closed 2022-09-02 by xdelu02